realsenseai/librealsense #5519: T265 docs - axes not matching expected?

msadowski, 2019-12-23:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265  | 
| Firmware Version                   | N/A | 
| Operating System & Version |   N/A  | 
| Kernel Version (Linux Only)    |  N/A                                     | 
| Platform                                 | N/A |
| SDK Version                            |  N/A }                          | 
| Language                            |  N/A                          | 
| Segment			|  {N/A                         | 

### Issue Description

Hi!

I was just re-looking at the T265 docs (nice to see them updated!) and I'm a bit confused about the camera transform showed in [the docs](https://github.com/IntelRealSense/librealsense/blob/master/doc/t265.md). Specifically I'm confused about this one:

[image]

Looking at the Z-axis the camera in this example is looking up (45 degree angle), however following the coordinate frames of the camera:

[image]

The x-axis of the pose-frame will be pointing "out of the screen". Following the right-hand rule tilting the camera like this should yield a positive angle but instead the config file included in the documentation mentions a negative angle. Am I missing something here?

neilyoung, 2019-12-23:
They are fast. It is fixed :)

neilyoung, 2019-12-23:
Although your image wouldn't  be incorrect at all. Your robot would be good too starring into the air. You would be surprised, how good a tracking along the ceiling works :)

msadowski, 2019-12-23:
@neilyoung I don't see the fix applied yet and I wouldn't mind the camera pointing upwards at all! I would only prefer that the drawing closely matches the config file as I think this will clear a lot of confusion for others.

neilyoung, 2019-12-23:
Sorry, I thought you would have had a problem with the up-pointing camera.

neilyoung, 2019-12-23:
Other than that I think the negative angle is correct, since it describes the rotation of the robot body w.r.t. the camera reference frame. So while the camera is pitching by +45 degrees (rotated counterclockwise), the body of the robot rotates by -45 degrees, clockwise. Correct?

neilyoung, 2019-12-23:
Second attempt: Just tested with my cam. If I start the cam tilted up 45 degrees it correctly reports a pitch of 45 degrees. This would not reflect the expected pitch of the entire system, which is determined by the robot. Hence you have to subtract 45 degrees from all reported pose pitch values (if we understand pitch here to be the rotation around the x axis) in order to get the "real" pitch of your system. Makes sense?

msadowski, 2019-12-23:
Thanks a lot for taking the time to test it!

The positive 45 degree angle output when tilting up sounds correct - that's what I would expect from the system. I'm not so sure about negating the value though just to match the result. IMO if the docs mention "In this example, we take the above setup and add **a rotation around the camera's X-axis.**" then I'd hope that in the tilt up case the x-axis rotation will have negative value and the angle correction should happen in the camera itself.

If the rotation indeed needs to be negated for this setup to work correctly then I think the docs should be modified to mention it so that there is no need to second-guess the image vs the config file.

RealSenseCustomerSupport, 2020-01-07:
----------------------------------------------

Thanks for the doc recommendation msadowski. We'll look into making that clarification.

Will you be needing any further assistance with this ticket?

msadowski, 2020-01-07:
@RealSenseCustomerSupport if you can confirm that indeed for the sensor looking into the ceiling the rotation specified in the config file should be negative that would be superb.

RealSenseCustomerSupport, 2020-01-07:
----------------------------------------------

Yes the negative value is correct as neilyoung explained. The values define the final transformation matrix that transforms a point from the robot frame to the camera frame. So a positive rotation of the camera results in a negative "rotation" of the robot.

neilyoung, 2020-01-07:
@RealSenseCustomerSupport Thanks for confirmation.

msadowski, 2020-01-07:
@RealSenseCustomerSupport many thanks for the clarification. I won't be needing any further assistance on this task, feel free to close it!

closed 2020-01-08 by RealSenseCustomerSupport