realsenseai/librealsense #5524: SIGBUS error on android during closing the camera and pipeline

kafan1986, 2019-12-23:
Required Info |  
-- | --
Camera Model | { D415}
Firmware Version | {05.11.15.00}
Operating System & Version | {Android}
Platform | {Arm-Odroid}
SDK Version | { 2.31.0 }
Language | {C++/Java }
Segment | {others }

I have received a new error after I opened and closed the camera 2-3 time. This happened after closing the camera. I can not pinpoint which line of java code threw this error exactly. Tombstone log is attached.

[tombstone_10.txt](https://github.com/IntelRealSense/librealsense/files/3995854/tombstone_10.txt)

```
2019-12-23 23:30:47.682 5093-5093/com.synapticsense.retailapp:UnityKillsMe I/librs: The user was holding on to 2 frames after stream 0x0x759ad5d620 stopped
2019-12-23 23:30:47.718 5093-5149/com.synapticsense.retailapp:UnityKillsMe I/librs: USB pipe 132 reset successfully
2019-12-23 23:30:47.762 5093-5093/com.synapticsense.retailapp:UnityKillsMe E/CRASH: signal 7 (SIGBUS), code 1 (BUS_ADRALN), fault addr ffffffffffffffff
2019-12-23 23:30:47.762 5093-5093/com.synapticsense.retailapp:UnityKillsMe E/CRASH: *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
2019-12-23 23:30:47.762 5093-5093/com.synapticsense.retailapp:UnityKillsMe E/CRASH: Build fingerprint: 'ODROID/odroidn2/odroidn2:9/PQ3A.190801.002/20191204:eng/test-keys'
2019-12-23 23:30:47.762 5093-5093/com.synapticsense.retailapp:UnityKillsMe E/CRASH: Revision: '0'
2019-12-23 23:30:47.763 5093-5093/com.synapticsense.retailapp:UnityKillsMe E/CRASH: pid: 5093, tid: 5093, name: pp:UnityKillsMe  >>> com.synapticsense.retailapp:UnityKillsMe <<<
2019-12-23 23:30:47.763 5093-5093/com.synapticsense.retailapp:UnityKillsMe E/CRASH:     x0   000000759924cf90  x1   0000000000000000  x2   000000759ca00000  x3   000000759ca97a40
2019-12-23 23:30:47.763 5093-5093/com.synapticsense.retailapp:UnityKillsMe E/CRASH:     x4   0000000000000000  x5   0000000000000000  x6   00000076408f5000  x7   0000000000030028
2019-12-23 23:30:47.763 5093-5093/com.synapticsense.retailapp:UnityKillsMe E/CRASH:     x8   ffffffffffffffff  x9   22acdb5d43ba38eb  x10  0000000000000001  x11  0000000000000000
2019-12-23 23:30:47.763 5093-5093/com.synapticsense.retailapp:UnityKillsMe E/CRASH:     x12  0000000000000001  x13  00000000000000c0  x14  0000000000000000  x15  000000000000007e
2019-12-23 23:30:47.763 5093-5093/com.synapticsense.retailapp:UnityKillsMe E/CRASH:     x16  000000759d7e23a0  x17  000000759d4a0300  x18  00000075bb08fdc0  x19  000000759adb15d0
2019-12-23 23:30:47.763 5093-5093/com.synapticsense.retailapp:UnityKillsMe E/CRASH:     x20  000000759c328d80  x21  00000075bae0dc00  x22  0000007fffbe94f0  x23  00000075a13d6a3b
2019-12-23 23:30:47.763 5093-5093/com.synapticsense.retailapp:UnityKillsMe E/CRASH:     x24  0000000000000008  x25  00000076409075e0  x26  00000076409075e0  x27  0000000000000002
2019-12-23 23:30:47.763 5093-5093/com.synapticsense.retailapp:UnityKillsMe E/CRASH:     x28  0000007fffbe9220  x29  0000007fffbe8e60  x30  000000759d4a031c
2019-12-23 23:30:47.763 5093-5093/com.synapticsense.retailapp:UnityKillsMe E/CRASH:     sp   0000007fffbe8e60  pc   ffffffffffffffff  pstate 0000000060000000
2019-12-23 23:30:47.763 5093-5093/com.synapticsense.retailapp:UnityKillsMe E/CRASH: backtrace:
2019-12-23 23:30:47.812 5093-5093/com.synapticsense.retailapp:UnityKillsMe E/CRASH: 	#00  pc 00000000000006bc  [vdso] ()
2019-12-23 23:30:47.812 5093-5093/com.synapticsense.retailapp:UnityKillsMe E/CRASH: 	#01  pc fffffffffffffffb  <unknown/absolute> ()
2019-12-23 23:30:47.812 5093-5093/com.synapticsense.retailapp:UnityKillsMe E/CRASH: chained_signal_handler 622 got 7
    
    --------- beginning of crash
2019-12-23 23:30:47.813 5093-5093/com.synapticsense.retailapp:UnityKillsMe A/libc: Fatal signal 7 (SIGBUS), code 1 (BUS_ADRALN), fault addr 0xffffffffffffffff in tid 5093 (pp:UnityKillsMe), pid 5093 (pp:UnityKillsMe)
2019-12-23 23:30:47.908 5291-5291/? I/crash_dump64: obtaining output fd from tombstoned, type: kDebuggerdTombstone
2019-12-23 23:30:47.909 2963-2963/? I//system/bin/tombstoned: received crash request for pid 5093
2019-12-23 23:30:47.910 5291-5291/? I/crash_dump64: performing dump of process 5093 (target tid = 5093)
2019-12-23 23:30:47.921 5291-5291/? A/DEBUG: *** *** *** *** *** *** *** *** *** *** *** *** *** *** *** ***
2019-12-23 23:30:47.922 5291-5291/? A/DEBUG: Build fingerprint: 'ODROID/odroidn2/odroidn2:9/PQ3A.190801.002/20191204:eng/test-keys'
2019-12-23 23:30:47.922 5291-5291/? A/DEBUG: Revision: '0'
2019-12-23 23:30:47.922 5291-5291/? A/DEBUG: ABI: 'arm64'
2019-12-23 23:30:47.922 5291-5291/? A/DEBUG: pid: 5093, tid: 5093, name: pp:UnityKillsMe  >>> com.synapticsense.retailapp:UnityKillsMe <<<
2019-12-23 23:30:47.922 5291-5291/? A/DEBUG: signal 7 (SIGBUS), code 1 (BUS_ADRALN), fault addr 0xffffffffffffffff
2019-12-23 23:30:47.922 5291-5291/? A/DEBUG:     x0  000000759924cf90  x1  0000000000000000  x2  000000759ca00000  x3  000000759ca97a40
2019-12-23 23:30:47.922 5291-5291/? A/DEBUG:     x4  0000000000000000  x5  0000000000000000  x6  00000076408f5000  x7  0000000000030028
2019-12-23 23:30:47.922 5291-5291/? A/DEBUG:     x8  ffffffffffffffff  x9  22acdb5d43ba38eb  x10 0000000000000001  x11 0000000000000000
2019-12-23 23:30:47.922 5291-5291/? A/DEBUG:     x12 0000000000000001  x13 00000000000000c0  x14 0000000000000000  x15 000000000000007e
2019-12-23 23:30:47.922 5291-5291/? A/DEBUG:     x16 000000759d7e23a0  x17 000000759d4a0300  x18 00000075bb08fdc0  x19 000000759adb15d0
2019-12-23 23:30:47.922 5291-5291/? A/DEBUG:     x20 000000759c328d80  x21 00000075bae0dc00  x22 0000007fffbe94f0  x23 00000075a13d6a3b
2019-12-23 23:30:47.922 5291-5291/? A/DEBUG:     x24 0000000000000008  x25 00000076409075e0  x26 00000076409075e0  x27 0000000000000002
2019-12-23 23:30:47.922 5291-5291/? A/DEBUG:     x28 0000007fffbe9220  x29 0000007fffbe8e60
2019-12-23 23:30:47.922 5291-5291/? A/DEBUG:     sp  0000007fffbe8e60  lr  000000759d4a031c  pc  ffffffffffffffff
2019-12-23 23:30:48.049 3165-3179/? W/ActivityManager: Activity pause timeout for ActivityRecord{4d7c957 u0 com.synapticsense.retailapp/.activities.MainActivity t76 f}
2019-12-23 23:30:48.051 3441-3441/com.synapticsense.retailapp W/ActivityThread: handleWindowVisibility: no activity for token android.os.BinderProxy@b624164
2019-12-23 23:30:48.078 3441-3441/com.synapticsense.retailapp D/MRApp_1.1 :SettingsActivity: Start Of onCreate
2019-12-23 23:30:48.150 5291-5291/? A/DEBUG: backtrace:
2019-12-23 23:30:48.150 5291-5291/? A/DEBUG:     #00 pc ffffffffffffffff  <unknown>
2019-12-23 23:30:48.150 5291-5291/? A/DEBUG:     #01 pc 00000000006ec318  /data/app/com.synapticsense.retailapp-Dc-Ja5FzZgAaaCnSr1I_Og==/lib/arm64/librealsense2.so (librealsense::frame_holder::~frame_holder()+24)
2019-12-23 23:30:48.150 5291-5291/? A/DEBUG:     #02 pc 00000000006094ac  /data/app/com.synapticsense.retailapp-Dc-Ja5FzZgAaaCnSr1I_Og==/lib/arm64/librealsense2.so (std::__ndk1::__tree<std::__ndk1::__value_type<int, librealsense::frame_holder>, std::__ndk1::__map_value_compare<int, std::__ndk1::__value_type<int, librealsense::frame_holder>, std::__ndk1::less<int>, true>, std::__ndk1::allocator<std::__ndk1::__value_type<int, librealsense::frame_holder>>>::destroy(std::__ndk1::__tree_node<std::__ndk1::__value_type<int, librealsense::frame_holder>, void
2019-12-23 23:30:48.150 5291-5291/? A/DEBUG:     #03 pc 00000000006094a4  /data/app/com.synapticsense.retailapp-Dc-Ja5FzZgAaaCnSr1I_Og==/lib/arm64/librealsense2.so (std::__ndk1::__tree<std::__ndk1::__value_type<int, librealsense::frame_holder>, std::__ndk1::__map_value_compare<int, std::__ndk1::__value_type<int, librealsense::frame_holder>, std::__ndk1::less<int>, true>, std::__ndk1::allocator<std::__ndk1::__value_type<int, librealsense::frame_holder>>>::destroy(std::__ndk1::__tree_node<std::__ndk1::__value_type<int, librealsense::frame_holder>, void
2019-12-23 23:30:48.150 5291-5291/? A/DEBUG:     #04 pc 0000000000609b94  /data/app/com.synapticsense.retailapp-Dc-Ja5FzZgAaaCnSr1I_Og==/lib/arm64/librealsense2.so (librealsense::pipeline::aggregator::~aggregator()+148)
2019-12-23 23:30:48.150 5291-5291/? A/DEBUG:     #05 pc 0000000000609398  /data/app/com.synapticsense.retailapp-Dc-Ja5FzZgAaaCnSr1I_Og==/lib/arm64/librealsense2.so (librealsense::pipeline::aggregator::~aggregator()+24)
2019-12-23 23:30:48.150 5291-5291/? A/DEBUG:     #06 pc 00000000005f85a4  /data/app/com.synapticsense.retailapp-Dc-Ja5FzZgAaaCnSr1I_Og==/lib/arm64/librealsense2.so (librealsense::pipeline::pipeline::~pipeline()+128)
[... cut ...]
2019-12-23 23:30:48.152 5291-5291/? A/DEBUG:     #138 pc 00000000000b1ef4  /system/lib64/libandroid_runtime.so (_JNIEnv::CallStaticVoidMethod(_jclass*, _jmethodID*, ...)+120)
2019-12-23 23:30:48.152 5291-5291/? A/DEBUG:     #139 pc 00000000000b4878  /system/lib64/libandroid_runtime.so (android::AndroidRuntime::start(char const*, android::Vector<android::String8> const&, bool)+756)
2019-12-23 23:30:48.152 5291-5291/? A/DEBUG:     #140 pc 000000000000219c  /system/bin/app_process64 (main+1200)
2019-12-23 23:30:48.152 5291-5291/? A/DEBUG:     #141 pc 00000000000a9804  /system/lib64/libc.so (__libc_init+88)
```

RealSenseCustomerSupport, 2020-01-03:
----------------------------------------------

@kafan1986 Did you get this issue with SDK example? If it only happens with your own code, could you provide your code snippet? Thanks!

kafan1986, 2020-01-03:
> @kafan1986 Did you get this issue with SDK example? If it only happens with your own code, could you provide your code snippet? Thanks!

I have made many changes and improved the situation. I have not seen the error again. Currently the issue plaguing the Android platform is "Out of Memory" issue and "Colour Stream" getting stuck issue. Happens within 30-45 mins on 1080p at 15 FPS and within 3-4 hours at 720p at 15 FPS. Only solution that recovers it a system reboot. 

Probably related to **(DSO-13539) - [Android] Camera disconnected after streaming some duration with Android Camera Sample** May be you can escalate the issue and expedite a solution, and in the meanwhile update us regarding the same.

I will be closing this issue in the meanwhile.

closed 2020-01-03 by kafan1986