realsenseai/librealsense #5527: T265: Enumeration after boot on Raspberry pi 4

zainmehdi, 2019-12-24:
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Operating System & Version |  Raspbian Buster  |                                        | 
| Platform                                 | Raspberry Pi 4 |
| SDK Version                            | 2.31                        | 

### Issue Description
Can anyone please provide any update on this issue. It seems like the camera is not detected after initial boot and one has to plug out and plug in to make it work. This is a serious problem that prevents this camera from being used in any of my products. Customers certainly cant plug out and plug in every time they want to use this device. I am really hoping Engineering team can provide some workaround for this issue.

beyond21299, 2019-12-27:
Do you make the T265 camera work in Pi4? I'm trying to install sdk2.31 ,but not work.

zainmehdi, 2019-12-27:
@beyond21299 Yes It works. What OS are you using ?

beyond21299, 2019-12-27:
> @beyond21299 Yes It works. What OS are you using ?

I'm sorry.I forget take photo.  I'm trying Pi 3B+ with   Linux raspberrypi 4.19.57-v7+.

beyond21299, 2019-12-27:
I'll try again today, hoping to repeat yesterday's situation. I remember the error when I installed libusb

zainmehdi, 2019-12-27:
I have tried it with RPi 4,RPi 3B+ , RPi Zero W, Banana Pi M2z, Nanao pi Neo2, Nano Pi Neo 4 and it works on all of them.

beyond21299, 2019-12-27:
> I have tried it with RPi 4,RPi 3B+ , RPi Zero W, Banana Pi M2z, Nanao pi Neo2, Nano Pi Neo 4 and it works on all of them.

what OS are you using in RPi 3B+?

zainmehdi, 2019-12-27:
I used Raspbian buster, But I also tried with Ubuntu mate

beyond21299, 2019-12-27:
> I used Raspbian buster, But I also tried with Ubuntu mate

Yes,I can use Raspbian buster OS, thank you ,I will try later

beyond21299, 2019-12-27:
> I used Raspbian buster, But I also tried with Ubuntu mate. If you dont have a strict requirement of using Other linux distro I would recommend you to have Raspbian buster and try with these instructions https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md. It should work.
when I install realsenseSDK2.0 follow this URL(https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_raspbian.md) on my RPi 3B+. there is errors like this.  Is **libdrm-***-dbg** necessary?

sudo apt-get install -y libdrm-amdgpu1 libdrm-amdgpu1-dbg libdrm-dev libdrm-exynos1 libdrm-exynos1-dbg libdrm-freedreno1 libdrm-freedreno1-dbg libdrm-nouveau2 libdrm-nouveau2-dbg libdrm-omap1 libdrm-omap1-dbg libdrm-radeon1 libdrm-radeon1-dbg libdrm-tegra0 libdrm-tegra0-dbg libdrm2 libdrm2-dbg
正在读取软件包列表... 完成
正在分析软件包的依赖关系树       
正在读取状态信息... 完成       
E: con't location package libdrm-amdgpu1-dbg
E: 无法定位软件包 libdrm-exynos1-dbg
E: 无法定位软件包 libdrm-freedreno1-dbg
E: 无法定位软件包 libdrm-nouveau2-dbg
E: 无法定位软件包 libdrm-omap1-dbg
E: 无法定位软件包 libdrm-radeon1-dbg
E: 无法定位软件包 libdrm-tegra0-dbg
E: 无法定位软件包 libdrm2-dbg

zainmehdi, 2019-12-27:
try this

 `sudo apt-get install -y libdrm-amdgpu1 libdrm-dev libdrm-exynos1 libdrm-freedreno1 libdrm-nouveau2 libdrm-omap1 libdrm-radeon1 libdrm-tegra0 libdrm2 `

zainmehdi, 2019-12-27:
and skip cmake protobuf tbb and python part if you dont need additional features. They take a very long time to compile. Make sure you have cmake installed from apt manager. Otherwise you can install it using `sudo apt-get install cmake`

beyond21299, 2019-12-27:
> and skip cmake protobuf tbb opencv and python part if you dont need additional features. They take a very long time to compile. Make sure you have cmake installed from apt manager. Otherwise you can install it using `sudo apt-get install cmake`

Thanks, I might want to try installing opencv as this instructed, because my code might need to use AprilTag later.

zainmehdi, 2019-12-27:
> Thanks, I might want to try installing opencv as this instructed, because my code might need to use AprilTag later.

Sure go ahead with that.

beyond21299, 2019-12-27:
> Sure go ahead with that.

Maybe I can install opencv later by myself . Now I compile RealsenseSDK. when I use this 'cmake .. -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=Release -DFORCE_LIBUVC=true', Is it take a long time during '--Building whit TM2'?

zainmehdi, 2019-12-27:
> Maybe I can install opencv later by myself . Now I compile RealsenseSDK. when I use this 'cmake .. -DBUILD_EXAMPLES=true -DCMAKE_BUILD_TYPE=Release -DFORCE_LIBUVC=true', Is it take a long time [...]

Its downloading files it needs for TM2. Please make sure your internet connection is working fine. Else it will try to build without TM2 which should also go well.

beyond21299, 2019-12-27:
> Its downloading files it needs for TM2. Please make sure your internet connection is working fine. Else it will try to build without TM2 which should also go well.

OK,The Internet connection is good. I'll wait a minute.

beyond21299, 2019-12-27:
[error.txt](https://github.com/IntelRealSense/librealsense/files/4004555/error.txt)
@zainmehdi I'm sorry. Can you help me analyze what's problem from my error

beyond21299, 2019-12-27:
this is the complete error output .
[error.txt](https://github.com/IntelRealSense/librealsense/files/4004562/error.txt)

zainmehdi, 2019-12-27:
> this is the complete error output .
> [error.txt](https://github.com/IntelRealSense/librealsense/files/4004562/error.txt)

https://github.com/IntelRealSense/librealsense/issues/4833 try this fix

radfordi, 2020-01-10:
Hi @zainmehdi, if you happen to have a Raspberry Pi 4**B**,  would you try powering the USB port, that the T265 is connect to, down and back up using [uhubctl](https://github.com/mvp/uhubctl) after the first boot?  We have found this to work for other devices.

radfordi, 2020-01-10:
@beyond21299, it looks like the firmware file that is downloaded has the wrong SHA1 hash.
```text
  file DOWNLOAD HASH mismatch

    for file: [/home/pi/librealsense/build/third-party/libtm/fw/target-0.2.0.879.mvcmd]
      expected hash: [b7b722b785fcf5f8c31877173c8ba02fed120e03]
        actual hash: [a2c2f80082d80e0842402ec7ed7ece7faff49449]
             status: [56;"Failure when receiving data from the peer"]
```
Try removing that file and trying again.   If you have a clean build directory and your network access is working, then the download should match the expected hash.

zainmehdi, 2020-01-11:
> Hi @zainmehdi, if you happen to have a Raspberry Pi 4**B**, would you try powering the USB port, that the T265 is connect to, down and back up using [uhubctl](https://github.com/mvp/uhubctl) after the first boot? We have found this to work for other devices.

Thanks for the update. I will try and update soon.

m4xr1sk, 2020-02-04:
uhubctl doesn't work for me with the Raspberry Pi 4B
it's always detected as:
Bus 001 Device 009: ID 03e7:2150 Intel Myriad VPU [Movidius Neural Compute Stick]
I have to manually unplug and re-plug the usb cable
how is that possibile?

radfordi, 2020-02-04:
Hi @m4xr1sk,

It looks like the [Raspberry Pi 4B is supported by `uhubctrl`](https://github.com/mvp/uhubctl#compatible-usb-hubs).  Have you upgraded your firmware as suggested [here](https://github.com/mvp/uhubctl#raspberry-pi-4b) to enable support for power cycling USB ports?

m4xr1sk, 2020-02-05:
the firmware is upgraded. I'm waiting for an external usb hub to try with it

radfordi, 2020-02-05:
Just to be clear, should see the following device in both cases.
> Bus 001 Device 009: ID 03e7:2150 Intel Myriad VPU
The difference will be in the output of this command.
```sh
$ lsusb -d 03e7:2150 -v | grep bcdUSB
2.0
```
When the device needs to be reset, it should in USB 1.0 mode, and it will be in USB 2.0 mode after the device is power cycled.

m4xr1sk, 2020-02-07:
that's the version of the bootloader of the RPI4:
[image]

Here the output of the lsusb after connecting the camera:
```
pi@raspberrypi:~/uhubctl $ lsusb
Bus 002 Device 010: ID 8087:0b37 Intel Corp. 
Bus 002 Device 005: ID 2109:8110 VIA Labs, Inc. Hub
Bus 002 Device 003: ID 8087:0b37 Intel Corp. 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 022: ID 2109:2811 VIA Labs, Inc. Hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```
in the case above the camera is powered and working correctly with rs-pose
[image]

if I power off the hub with the command:
`sudo ./uhubctl -l 2-2 -a 0`
the usb meter goes off, as you see in this image
[image]

with the command:
`sudo ./uhubctl -l 2-2 -a 1`
the hub power on again the meter and the camera:
[image]

(note that in that case the current consumption is different)
but the camera is not working anymore.

That's the output of the rs-pose:
```
pi@raspberrypi:~/uhubctl $ rs-pose
RealSense error calling rs2_pipeline_start_with_config(pipe:0x34f1a0, config:0x3376b8):
    No device connected
```

Here the version of the library:
[image]

radfordi, 2020-02-07:
Hi @m4xr1sk,

Thanks for the nice pictures!    Could you provide the output of this command after you use `uhubctl` to power the device back on?
```
lsusb -d 03e7:2150  | grep bcdUSB
```

m4xr1sk, 2020-02-08:
just after connecting the camera:
```
pi@raspberrypi:~ $ lsusb -d 03e7:2150 -v | grep bcdUSB
can't get debug descriptor: Resource temporarily unavailable
  bcdUSB               2.00
  bcdUSB               2.00
```

power off:
```
pi@raspberrypi:~ $ sudo uhubctl/uhubctl -l 2-2 -a 0
Current status for hub 2-2 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 0203 power 5gbps U0 enable connect [8087:0b37 Intel(R) Corporation Intel(R) RealSense(TM) Tracking Camera T265 925122111036]
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Sent power off request
New status for hub 2-2 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 0080 off
  Port 2: 0080 off
  Port 3: 0080 off
  Port 4: 0080 off
Current status for hub 1-1.2 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0503 power highspeed enable connect []
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
Sent power off request
New status for hub 1-1.2 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0000 off
  Port 2: 0000 off
  Port 3: 0000 off
  Port 4: 0000 off
```

power on:
```
pi@raspberrypi:~ $ sudo uhubctl/uhubctl -l 2-2 -a 1
Current status for hub 2-2 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 0080 off
  Port 2: 0080 off
  Port 3: 0080 off
  Port 4: 0080 off
Sent power on request
New status for hub 2-2 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1.2 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0000 off
  Port 2: 0000 off
  Port 3: 0000 off
  Port 4: 0000 off
Sent power on request
New status for hub 1-1.2 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0100 power
```

and that's the output of the uhubctl:
```
pi@raspberrypi:~ $ lsusb -d 03e7:2150 -v | grep bcdUSB
can't get debug descriptor: Resource temporarily unavailable
  bcdUSB               2.00
  bcdUSB               2.00
```

everything seems correct but if I run rs-pose it doesn't output anything (just the error).
That's the output of the rs-fw-update:
```
pi@raspberrypi:~ $ rs-fw-update -l

connected devices:
 08/02 09:38:44,365 ERROR [3069415440] (tm-device.cpp:2036) Bulk request error RS2_USB_STATUS_IO
 08/02 09:38:44,365 ERROR [3069415440] (types.h:304) Unable to communicate with device
RealSense error calling rs2_create_device(info_list:0x85c5c8, index:0):
    Unable to communicate with device
```

mvp, 2020-02-09:
uhubctl author here.

@m4xr1sk , does your camera re-enumerate if you unplug/replug USB cable at this point?

If it does, can you try adding -R to uhubctl power on command and see if it helps?

m4xr1sk, 2020-02-10:
Hi mvp,
thanks for your suggestion.
If I unplug/replug the USB it always works.
Now I'm able to work with the camera with this script:

```
#!/bin/sh
sudo ./uhubctl -e -l 2-2   -a 0
sudo ./uhubctl -e -l 1-1.2 -a 0
sleep .5
sudo ./uhubctl -e -l 1-1.2 -a 1 -R
sudo ./uhubctl -e -l 2-2   -a 1 -R
sleep .5
rs-pose
```

If I don't use -e or -R it doesn't work.
Sleep commands are optional (rs-pose is the program that I use to test the camera).
Also, the camera works only if connected to the amazon hub.
If I connect the camera directly to the RPI it doesn't work.

At the moment the above could be an acceptable solution.
It could be nice to understand why it works only with the amazon hub and not directly connected to the RPI. Let me know if I can help with the output of something

mvp, 2020-02-10:
-R is an attempt to fully reset USB bus, should be similar to physical replugging and re-enumeration.

It should work when directly plugged in RPi4B, but you may need to reset all 4 ports with `sudo uhubctl -l 2 -a cycle -R`.

As for AmazonBasics hub and necessity to use -e, that is not normal. Can you please share output of `sudo uhubctl -l 2-2`? It should list both 2-2 and 1-1.2, and command `sudo uhubctl -l 2-2 -a cycle -R` should work.

yousseb, 2020-02-11:
Based on uhubctl by @mvp, I created the repo:

https://github.com/yousseb/t265_reset

which basically recycles the specific hub port that a T265 is connected to.  
@mvp : Thanks for uhubctl

mvp, 2020-02-11:
@yousseb , I would recommend just calling uhubctl instead. It will get updated over time

radfordi, 2020-02-11:
@mvp, we appreciate your point, but our goal with this repo is to eventually integrate it into `librealsense` so that the resetting would happen automatically for most users and would work on platforms, like Android and Windows, where we don't use `libusb`.  We are fully aware that this solution would always be less complete than just using `uhubctl` directly.

mvp, 2020-02-11:
@radfordi , it will not work on Windows. It doesn't have ability to send control message directly to the hub

radfordi, 2020-02-11:
Thanks, @mvp, for the heads up.  No need to test there now. :)

mvp, 2020-02-11:
@yousseb, please change license on https://github.com/yousseb/t265_reset to GPL2 then. it is based mostly on uhubctl code, which is GPL2.

m4xr1sk, 2020-02-12:
> -R is an attempt to fully reset USB bus, should be similar to physical replugging and re-enumeration.
>
> It should work when directly plugged in RPi4B, but you may need to reset all 4 ports with `sudo [...]

Directly connected to the RPI it doesn't work.
out1.txt is the output of sudo uhubctl -l 2 -a cycle -R
[out1.txt](https://github.com/IntelRealSense/librealsense/files/4190114/out1.txt)

Also with the param "cycle" and connected to the amazon hub it doesn't work.
out2.txt is the output of sudo uhubctl -l 2-2 -a cycle -R
[out2.txt](https://github.com/IntelRealSense/librealsense/files/4190119/out2.txt)

here the output of sudo uhubctl -l 2-2
```
pi@raspberrypi:~/uhubctl $ sudo ./uhubctl -l 2-2 
Current status for hub 2-2 [2109:8110 VIA Labs, Inc. USB3.0 Hub, USB 3.00, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Current status for hub 1-1.2 [2109:2811 VIA Labs, Inc. USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0503 power highspeed enable connect [03e7:2150 Movidius Ltd. Movidius MA2X5X 03e72150]
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0103 power enable connect [0403:6001 FTDI FT232R USB UART AO008YRK]
```

yousseb, 2020-02-12:
@mvp It’s already GPL2 (LICENSE file). What am I missing?

mvp, 2020-02-12:
@yousseb , I just looked at first line of `main.cpp`:

https://github.com/yousseb/t265_reset/blob/95ba890fb5fd56b0ebe022ef86a96e5c431a3a0c/main.cpp#L1

and it says:

    // Copyright(c) 2020 Intel Corporation. All Rights Reserved.

I understand that you added LICENSE with GPL2 just yesterday. But perhaps you should be consistent with your licensing information.

mvp, 2020-02-12:
@m4xr1sk , it seems that the only difference between your approach 

    sudo ./uhubctl -e -l 2-2   -a 0
    sudo ./uhubctl -e -l 1-1.2 -a 0
    sleep .5
    sudo ./uhubctl -e -l 1-1.2 -a 1 -R
    sudo ./uhubctl -e -l 2-2   -a 1 -R
 
and if you call `sudo uhubctl -l 2-2 -a cycle -R` would do the same, but in slightly different order, it would be equivalent to:

    sudo ./uhubctl -e -l 2-2   -a 0
    sudo ./uhubctl -e -l 1-1.2 -a 0
    sleep 2
    sudo ./uhubctl -e -l 2-2   -a 1 -R    # note order of USB2/USB3 is swapped
    sudo ./uhubctl -e -l 1-1.2 -a 1 -R

I will think about possibly swapping USB2/USB3 turn on order in uhubctl...

m4xr1sk, 2020-02-15:
The Intel T265 stop working (using rs-pose) after some minutes if connected to the Amazon Hub. If I launch again rs-pose it works.
Directly connected to the Raspberry PI 4: no problems.
How I can investigate the problem? Any suggestion?

radfordi, 2020-02-17:
Bad cable?  Not enough power for both the hub and the T265?  Are you using an external power supply for the hub?

m4xr1sk, 2020-02-17:
Same conditions but back to SDK v2.31.0 and now it seems to work.
Anyway, the cable is ok, I can't use the external power supply for the hub in the final application (like mostly moving solutions) and the current drawn by the camera is only 250mA.

colin-broderick, 2020-03-11:
@mvp I could have sworn I had this working the other day but now I can't reproduce it. I ran `sudo uhubctl -l 2 -a cycle -R` as suggested, and the output is

```
Current status for hub 2 [1d6b:0003 Linux 4.19.97-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Sent power off request
New status for hub 2 [1d6b:0003 Linux 4.19.97-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports]
  Port 1: 0080 off
  Port 2: 0080 off
  Port 3: 0080 off
  Port 4: 0080 off
Current status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0503 power highspeed enable connect [03e7:2150 Movidius Ltd. Movidius MA2X5X 03e72150]
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0103 power enable connect [046d:c534 Logitech USB Receiver]
Sent power off request
New status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0000 off
  Port 2: 0000 off
  Port 3: 0000 off
  Port 4: 0000 off
Current status for hub 2 [1d6b:0003 Linux 4.19.97-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports]
  Port 1: 0080 off
  Port 2: 0080 off
  Port 3: 0080 off
  Port 4: 0080 off
Sent power on request
New status for hub 2 [1d6b:0003 Linux 4.19.97-v7l+ xhci-hcd xHCI Host Controller 0000:01:00.0, USB 3.00, 4 ports]
  Port 1: 02a0 power 5gbps Rx.Detect
  Port 2: 02a0 power 5gbps Rx.Detect
  Port 3: 02a0 power 5gbps Rx.Detect
  Port 4: 02a0 power 5gbps Rx.Detect
Resetting hub...
Reset failed!
: Is a directory
Current status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0000 off
  Port 2: 0000 off
  Port 3: 0000 off
  Port 4: 0101 power connect [046d:c534]
Sent power on request
New status for hub 1-1 [2109:3431 USB2.0 Hub, USB 2.10, 4 ports]
  Port 1: 0100 power
  Port 2: 0100 power
  Port 3: 0100 power
  Port 4: 0101 power connect [046d:c534]
Resetting hub...
Reset successful!

```

After which the RealSense does not work. Like others, it will always work by physically replugging but that's not an option.

I'm not using a smart hub. Can you clarify whether this resets all USB busses or ONLY works for external hubs?

rmackay9, 2020-03-18:
I can confirm that installing and building uhubctl ([as described here](https://github.com/mvp/uhubctl#compiling)) and then running these two commands:

- sudo ./uhubctl -a cycle -l 1 -p 1-4
- sudo ./uhubctl -a cycle -l 2 -p 1-4

allowed me to connect to the T265.  I'm using RPI4 running Ubuntu 18.04

RealSenseSupport, 2020-10-26:
Hi,

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

closed 2020-10-30 by RealSenseSupport

yutaide, 2020-10-31:
I wrote how to reset hub to recognize T265 with uhubctl.
https://www.notion.so/laplust/Recognize-T265-with-uhubctl-1b36892a104e4375b37a7a22bdaa9eb4