realsenseai/librealsense #5528: How to adjust the exposure time？

ningfenger, 2019-12-24:
I'm using c++, could you tell me how to adjust the exposure time? Thank you very much.

dorodnic, 2019-12-24:
Something like:
```cpp
rs2::pipeline pipe; 
rs2::pipeline_profile selection = pipe.start();
rs2::device selected_device = selection.get_device();
auto depth_sensor = selected_device.first<rs2::depth_sensor>();

if (depth_sensor.supports(RS2_OPTION_EXPOSURE))
{
    auto range = depth_sensor.get_option_range(RS2_OPTION_EXPOSURE); 
    depth_sensor.set_option(RS2_OPTION_EXPOSURE, range.min); // set minimal allowed exposure 
}
```
(There is more info at [wiki/API-How-To](https://github.com/IntelRealSense/librealsense/wiki/API-How-To))
RealSense camera has several independent sensors, this snippet is setting exposure of the depth/IR sensor. To control exposure of the RGB sensor, you can find it in the list returned from `selected_device.query_sensors()`, and modify its exposure same as above.

ningfenger, 2019-12-25:
Thank you and Merry Christmas.
However, when I use opencv, it changes my channel.
`int main() {
    pipeline pipe;
    rs2::config cfg;
    cfg.enable_stream(RS2_STREAM_COLOR, 640, 480, RS2_FORMAT_BGR8, 30);
    cfg.enable_stream(RS2_STREAM_DEPTH, 640, 480, RS2_FORMAT_Z16, 30);
    pipeline_profile selection = pipe.start();
    device selected_device = selection.get_device();
    auto depth_sensor = selected_device.first<rs2::depth_sensor>();
    if (depth_sensor.supports(RS2_OPTION_EXPOSURE))
    {
        auto range = depth_sensor.get_option_range(RS2_OPTION_EXPOSURE);
        depth_sensor.set_option(RS2_OPTION_EXPOSURE, range.min); // set minimal allowed exposure
    }
    while (1){
        rs2::frameset data = pipe.wait_for_frames(); // Wait for next set of frames from the camera
        rs2::align align(rs2_stream::RS2_STREAM_COLOR); // align depth and rgb image
        rs2::frameset aligned_frm = align.process(data); // aligned depth to rgb

        rs2::frame color = aligned_frm.get_color_frame();
        rs2::frame depth = aligned_frm.get_depth_frame();
        Mat color_img(Size(640, 480), CV_8UC3, (void *) color.get_data(), Mat::AUTO_STEP);
        imshow("color",color_img);
        waitKey(1);
    }
}`
[image]

ningfenger, 2019-12-25:
What's more， how can I change back？

ningfenger, 2019-12-25:
Ah, I mean, I want to adjust the exposure time of the RGB camera.

RealSenseCustomerSupport, 2020-01-03:
----------------------------------------------

@ningfenger It's similar for RGB camera. You can change depth sensor to RGB sensor to adjust the RGB exposure time.

ningfenger, 2020-01-04:
Think you， I 'll try.

ningfenger, 2020-01-09:
Thank you！ I solved the problem。

closed 2020-01-09 by ningfenger