realsenseai/librealsense #5531: Nvidia Nano RealSense Install - E: Unable to correct problems, you have held broken packages.

tmichals, 2019-12-24:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T235 and D435 | 
| Firmware Version                   | JetPack 4.3 - L4T 32.3.1 | 
| Operating System & Version | Ubuntu 18.04 LTS | 
| Kernel Version (Linux Only)    |  Linux 4.9.140-tegra aarch64                                    | 
| Platform                                 | NVIDIA Jetson|
| SDK Version                            |    v2.23.0          | 

I just downloaded and Flashed my Nano with the latest release JetPack 4.3 and tried installing from
https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md  and am getting the attached error.

[Nvidia Nano RealSense SDK Install.txt](https://github.com/IntelRealSense/librealsense/files/3998423/Nvidia.Nano.RealSense.SDK.Install.txt)

ev-mp, 2019-12-24:
@tmichals hello, based on the attached log it seem that a 3rd-part package or its dependencies are broken:
>libgtk-3-dev but it is not installable

Try one of the Ubuntu method to rectify it, e.g:
```sh
sudo apt-get update –fix-missing
sudo dpkg –configure -a
sudo apt-get install -f
```

tmichals, 2019-12-24:
JetsonHacks must have fixed the library or I corrected something that I was doing wrong.   I used the latest JetsonHacks Realsense install and it worked.

closed 2019-12-24 by tmichals