realsenseai/librealsense #5532: D435 new firmware v5.12 PROBLEM

scottnortman, 2019-12-24:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435 | 
| Firmware Version                   | v5.12 | 
| Operating System & Version |   Ubuntu 16.04  | 
| Kernel Version (Linux Only)    |  4.15.0-72-generic                                         | 
| Platform                                 | Dell Inspiron 2330 i5 Desktop  |
| SDK Version                            |  2.31.0                          | 
| Language                            |  ROS + Python                          | 
| Segment			|  other                     | 

### Issue Description

Just upgraded to the latest v5.12 firmware for a brand new D435 camera.

This latest update seems to break multiple features of the D435 camera:

1) Unable to set/stream both raw IR images (only depth or IR1 is now showing up in realsense viewer)

2) Many available streaming modes that were previously allowed in the prior firmware version are no longer listed, such as IR1 and IR2 @ 1280 x 720 @ 15 fps; now I can only stream  IR1 @ 6 fps

I've attached the output of  

$ rs-enumerate-devices -c

Can I roll back to the prior FW version?

[rs_d435_fw_5_12.txt](https://github.com/IntelRealSense/librealsense/files/3998604/rs_d435_fw_5_12.txt)

dorodnic, 2019-12-24:
Hi @scottnortman 
I don't think this is an issue with camera firmware or the library. Please note:
`Usb Type Descriptor           : 	2.1`
Try reconnecting the camera. Even using USB3 port, the camera can be sometimes misidentified as USB2.

scottnortman, 2019-12-24:
Additional info:

When I instead connect the same D435 camera to my windows machine using Realsense Viewer 2.29.0, I am able to see the expected D435 streaming modes and available streams.

scottnortman, 2019-12-24:
@dorodnic 

Thanks for the fast response, you were correct!  All working now after I unplugged / replugged the camera.

closed 2019-12-24 by scottnortman