realsenseai/librealsense #5533: how do I set rs2_options in the python examples?

skywo1f, 2019-12-24:
Lets say I am trying to use the t265_example.py script to get my position information, but I want to enable the RS2_OPTION_ENABLE_RELOCALIZATION rs2_option. Where would I go about setting that option?

skywo1f, 2019-12-26:
Or if anyone can answer how to enable that rs2_option anywhere, I may be able to figure it out from there. But for now it looks like the only way is to change the src and recompile?

neilyoung, 2019-12-26:
Actually it is pretty easy to do, if you know how :)

Call before starting the pipeline:

```
pipeline = rs.pipeline()
config = rs.config()
device = config.resolve(pipeline).get_device()
pose_sensor = device.first_pose_sensor()
pose_sensor.set_option(rs.option.enable_map_relocalization, 1)

```

BTW: RS2_OPTION_ENABLE_RELOCALIZATION is `True` by default.

These are the other related defaults:

```
RS2_OPTION_ENABLE_POSE_JUMPING 1
RS2_OPTION_ENABLE_MAPPING 1
RS2_OPTION_ENABLE_RELOCALIZATION 1
RS2_OPTION_ENABLE_MAP_PRESERVATION 0

```

closed 2019-12-27 by skywo1f

zwn, 2020-03-11:
> These are the other related defaults:
> 
> ```
> RS2_OPTION_ENABLE_POSE_JUMPING 1
> RS2_OPTION_ENABLE_MAPPING 1
> RS2_OPTION_ENABLE_RELOCALIZATION 1
> RS2_OPTION_ENABLE_MAP_PRESERVATION 0
> ```

@neilyoung Where did you get this information? I do see 

https://github.com/IntelRealSense/librealsense/blob/83f952a4bd6b70d72459f66c7f67ddaba9d337a0/doc/t265.md#are-there-any-t265-specific-options

but the defaults are not mentioned there. From the naming it would seem that the default is "disabled" since it is called "enable_something".

neilyoung, 2020-03-11:
I think I just read them back after reset. But I m not thar sure anymore. Clarification appreciated

neilyoung, 2020-03-12:
@zwn Double checked it with my own application (https://neilyoung.serveblog.net): The only option I set there after reset is

`            self.__pose_sensor.set_option(rs.option.enable_map_preservation, 1)
`

So my app is mapping, I see relocalizations and pose jumps. Just the map is not kept between sessions, if I don't enable it.

All this make me convenient again, that the a.m. default settings are correct.