realsenseai/librealsense #5536: How can i align color frame and depth frame by method "poll_for_frames()"

zhangzhe1103, 2019-12-25:
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D435 } | 
| Firmware Version                   | ( 5.10.13 ) | 
| Operating System & Version |   { Linux (Ubuntu 16.04)  | 
| Kernel Version (Linux Only)    |  (4.15.0)                                         | 
| Platform                                 | PC |
| SDK Version                            |  { 2.19 }                          | 
| Language                            |  {python/ }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
<I can align color frame and depth frame by method "pipeline.wait_for_frames",now I want to use multiple cameras，how can i align color frame and depth frame by method "poll_for_frames()">

zhangzhe1103, 2019-12-25:
`
align_to = rs.stream.color
align = rs.align(align_to)
while True:
        frames = pipeline.wait_for_frames()
        # frames.get_depth_frame() is a 640x360 depth image
        
        # Align the depth frame to color frame
        aligned_frames = align.process(frames)
        
        # Get aligned frames
        aligned_depth_frame = aligned_frames.get_depth_frame() # aligned_depth_frame is a 640x480 depth image
        color_frame = aligned_frames.get_color_frame()
`
but i don't now how to align by "pipeline.poll_for_frames()"

ev-mp, 2019-12-26:
@zhangzhe1103 hello,
Using the `pipe.poll_for_frames` is probably not the best method for this use-case as it is designed to return every frame exactly one time. It means that based on your application `frame- consumption` speed the API call mat return only partial framesets (rdepth OR color), and it will require extra handling in the application code to achieve alignment and visualization.

There are at least two alternatives to consider:
- Synchronous API:
   - `pipe.try_wait_for_frames` - Use this API with timeout <=2-5 msec to produce full framesets while minimizing the main-thread blocking intervals.
- Asynchronous API:
  - Use `pipe.start(<user_callback>)`  to get temporally-synchronized depth+color framesets without impeding the main thread. Note that this will probably require to rewrite some parts of the user code to adjust to asynchronous data handling

closed 2020-01-03 by zhangzhe1103