realsenseai/librealsense #5537: How to make the python program of the camera suddenly disconnected to resume without closing it?

Dontla, 2019-12-25:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435| 
| Firmware Version                   | 05.11.06.250| 
| Operating System & Version |  Win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.25.0                     | 
| Language                            |  python3.6                           | 
| Segment			|  Robot                       | 

### Issue Description
<Describe your issue / question / feature request / etc..>
During the normal transmission of the camera, my on-board computer sometimes disconnects the camera due to very strong vibration, and the program will immediately interrupt and display an error:
[image]
When this happens, we have to restart the program. I would like to ask you howcan keep the camera running after encountering a sudden disconnection of USB due to vibration without restarting the program?
Best regard

dorodnic, 2019-12-25:
Hi @Dontla 
Pipeline object should absorb disconnects of up to 5sec and resume streaming. If it doesn't you can always `try ... except` the error and retry streaming. Finally, context object offers some callbacks to discover when camera was disconnected / reconnected.

RealSenseCustomerSupport, 2020-05-03:
----------------------------------------------

Hi.

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

closed 2020-05-11 by RealSenseCustomerSupport