realsenseai/librealsense #5538: Announcement:  Highlighting the autoVimation IP66 protective enclosures for RealSense D415 and D435

MartyG-RealSense, 2019-12-25:
​Hi everyone,

With the emergence of industrial rugged variants of RealSense 400 Series cameras, such as the LIPS Corp's LIPSedge AE400 and the FRAMOS D435e, I thought it would be a good opportunity to highlight to owners of the Intel RealSense D415 and D435 camera models the German company autoVimation's range of industrial IP66 rated enclosures for those cameras. Details can be found in the autoVimation PDF data sheet linked to below.

https://www.autovimation.com/en/downloads/datenblaetter-deutsch/368-chameleon-xs-realsense/

closed 2020-01-02 by MartyG-RealSense

masip85, 2023-02-23:
This enclosures are super over prized. More than the camera itself! Autovimation enclosures cost 350-400 dollars (depending on provider). Framos prize, with cameras, cost around 1000 dollars. AE400 cost that too. Is very annoying that these cases cost 2 to 4 times the cost of the camera.

MartyG-RealSense, 2023-02-23:
Hi @masip85  The new RealSense **D457** camera model has IP65 dust and water protection for $499 USD.  Although it is primarily designed for a high-speed **GMSL / FAKRA** connection, it can be used with a standard USB cable by moving a switch on the camera to USB mode and provides the USB performance and features of a D455 model in that mode. 

https://store.intelrealsense.com/buy-intel-realsense-depth-camera-d457.html