realsenseai/librealsense #5540: T265 drifting all the time

gerokatseros, 2019-12-26:
| | |
|---|---|
|**librealsense**|2.29.0 RELEASE|
|**OS**|Windows|
|**Name**|Intel RealSense T265|
|**Serial Number**|909212111150|
|**Firmware Version**|0.1.0.279|
|**Product Line**|T200|

I get: UNKNOWN in rs2_set_option(options:0000015B54667DC0, option:Enable Auto Exposure, value:0):
Failed set manual exposure mode
Then ... 
In the beggining the 2 fisheye cameras flicker a little and the pose goes crazy and even though camera is steady it seems to constantly move.

This happens in both Ubuntu and Windows. The drift is huge and constant. I have enabled mapping, relocalization Pose Jumping, Dynamic calibration. With the same settings it worked fine. It's only a 2 weeks camera

zainmehdi, 2019-12-27:
Did you try to set exposure manually ?

gerokatseros, 2020-01-04:
I get:
13:48:20.294 [5301] [E] Device-7830: FW crashed - got error in interrupt endpoint thread function: status = -4 (LIBUSB_ERROR_NO_DEVICE), actual = 0
13:48:20.294 [5302] [E] Device-7830: FW crashed - got error in stream endpoint thread function: status = -4 (LIBUSB_ERROR_NO_DEVICE), actual = 243712
13:48:20.494 [5216] [E] Device-7830: State [ACTIVE_STATE] got event [ON_ERROR] ==> [ERROR_STATE]
13:48:20.494 [5216] [E] Device-7830: Entered state [ERROR_STATE]
04/01 13:49:43,969 WARNING [139815705855744] (tm-device.cpp:1076) Dropped frame. Frame length is 0
 04/01 13:49:44,070 WARNING [139815705855744] (tm-device.cpp:1076) Dropped frame. Frame length is 0
 04/01 13:49:44,560 WARNING [139815705855744] (tm-device.cpp:1076) Dropped frame. Frame length is 0
 04/01 13:49:44,679 WARNING [139815705855744] (tm-device.cpp:1076) Dropped frame. Frame length is 0
 04/01 13:49:44,882 WARNING [139815705855744] (tm-device.cpp:1076) Dropped frame. Frame length is 0
 04/01 13:49:45,034 WARNING [139815705855744] (tm-device.cpp:1076) Dropped frame. Frame length is 0
 04/01 13:49:45,051 WARNING [139815705855744] (tm-device.cpp:1076) Dropped frame. Frame length is 0
 04/01 13:49:45,203 WARNING [139815705855744] (tm-device.cpp:1076) Dropped frame. Frame length is 0
 04/01 13:49:45,288 WARNING [139815705855744] (tm-device.cpp:1076) Dropped frame. Frame length is 0

RealSenseCustomerSupport, 2020-01-06:
----------------------------------------------

@gerokatseros

Can you provide let me know if this is happening when you're using the RealSense Viewer?  And is it happening with the latest librealsense 2.31.0 release version?

Within RealSense viewer, are you trying to switch controls while the device is streaming or prior to enabling the stream?  Are you attempting to stream all streams at the same time?  Is the device being connected to a USB3 port on your system?

gerokatseros, 2020-01-06:
yes , it is happening with Realsense viewer , I haven't tried something else. It happens even with D415 off and only  T265 on. What I noticed, is that if I disable the infrared cameras from sending images, then it doesn't happen. But before the latest update on the same USBs it was working fine.

RealSenseCustomerSupport, 2020-01-23:
----------------------------------------------

Thanks @gerokatseros for the info.

RealSenseSupport, 2020-09-17:
Thank you for highlighting the drift issue with T265. We have moved our focus to our next generation of products and consequently, we will not be addressing this issue in the T265.

gerokatseros, 2020-09-18:
This is so sad!!! so you mean that T265 doesn't work well but you will ignore the issue and support another project? I thought this was a serious product and paid much money for itl!!!

zainmehdi, 2020-09-18:
@gerokatseros Same here. I am sure there must be a very strong reason for a company like Intel to make this kind of move. I myself based my whole lot of products on this tracking camera and when you get to know that it wont be supported any further or maybe withdrawn from the market the whole plan gets disturbed. It was certainly something you dont expect from intel with 100s of people working on roadmaps and stuff.

gerokatseros, 2020-09-18:
It makes me wonder if it's worth investing in a name like Intel for reliable products or at least some support after this. Why should I buy the upcoming products if I feel that any flaws will not be supported properly? Paying 200$ for a product that doesn't work as advertised and then telling you that it will not be supported is like waist of money (at least). They should at least give a disscount for a future product that works better or even replace the bad T265 with the upcoming product

closed 2021-03-31 by RealSenseSupport