realsenseai/librealsense #5542: Point clouds has sharp fluctuations

Tom127, 2019-12-26:
@dorodnic @ev-mp 
hello, I want to test the accuracy of `(x, y, z,)` calculated by `rs.rs2_deproject_pixel_to_point`, let the camera face the wall in parallel to calculate the corresponding `(x, y, z)` and visualize it on `rviz`, but as shown in the video, `(x, y, z)` fluctuates sharply. I don't know why? Normal conditions should be relatively stable, aren't they? 
thanks for your help! Attach my code:

[projection_test.txt](https://github.com/IntelRealSense/librealsense/files/4002191/projection_test.txt)

[pointclouds.mp4.zip](https://github.com/IntelRealSense/librealsense/files/4002192/pointclouds.mp4.zip)

ev-mp, 2019-12-26:
@Tom127 hello,
It is not clear from the capture what device type is used, the range to the target and target's characteristics: texture, reflectivity, ambient conditions , so I can outline the general recommendations only- 
1. There are recommended setups (resolution/presets) per camera type tuned for accuracy. For instance the best results for D415 are achieved with 1280X720. For D435 you should use 848X480.
https://github.com/IntelRealSense/librealsense/wiki/D400-Series-Visual-Presets.
2. Apply [post-processing filters](https://github.com/IntelRealSense/librealsense/blob/master/doc/post-processing-filters.md), the `temporal` filter in particular

There is a significant body of depth noise improvement techniques and methods discussed, e.g. #1280, #1375, #2289, #3625.

A concised summary of the recommended readings was compiled by @MartyG-Realsense in [3625](https://github.com/IntelRealSense/librealsense/issues/3625#issuecomment-477980700)

ev-mp, 2020-01-06:
@Tom127 hello, was the information helpful ?
Please update

RealSenseCustomerSupport, 2020-01-06:
----------------------------------------------

Hi @Tom127,

Do you still need help with this question?

Thank you!

closed 2020-01-13 by RealSenseCustomerSupport