realsenseai/librealsense #5544: Saving image as Grey color instead of color image

lokeshgs87, 2019-12-26:
The following is the code which we have implemented to save the image, but the image is getting saved in grey color . Can you please let us know the changes to be done to get color image instead of grey image 

Thread mStreaming = new Thread() {
        @Override
        public void run() {
            String filePath = mUri.getPath().split(":")[1];
            try(Colorizer colorizer = new Colorizer()) {
                try (Config config = new Config()) {
                    config.enableDeviceFromFile(filePath);
                    try (Pipeline pipeline = new Pipeline()) {
                        try {
                            pipeline.start(config);
                            while (!mStreaming.isInterrupted()) {
                                try (FrameSet frames = pipeline.waitForFrames(1000)) {
                                    try(FrameSet align_frames = frames.applyFilter(mAlign)) {
                                        try (FrameSet processed = align_frames.applyFilter(colorizer)) {
                                            processed.foreach(new FrameCallback() {
                                                @Override
                                                public void onFrame(Frame frame) {
                                                    try {
                                                        if (frame.getProfile().getType() == StreamType.COLOR) {
                                                            VideoFrame depthFrame = frame.as(Extension.VIDEO_FRAME);

                                                            Log.e(TAG, "streaming: before  saving bitmap1 ");

                                                            Mat mDepth = new Mat(depthFrame.getHeight(), depthFrame.getWidth(), CV_16UC1);
                                                            Log.e(TAG, "streaming: before  saving bitmap2 ");

                                                            int size = (int) (mDepth.total() * mDepth.elemSize());
                                                            byte[] return_buff = new byte[size];
                                                            //  depthFrame.getData(return_buff)

                                                            //  byte[] depth_buff = new byte[depthFrame.getHeight() * depthFrame.getStride()];
                                                            depthFrame.getData(return_buff);
                                                            short[] shorts = new short[size / 2];
                                                            ByteBuffer.wrap(return_buff).order(ByteOrder.LITTLE_ENDIAN).asShortBuffer().get(shorts);

                                                            Log.e(TAG, "streaming: before  saving bitmap2.1 ");
                                                            mDepth.put(0, 0, shorts);
                                                            Log.e(TAG, "streaming: before  saving bitmap3 ");

                                                            Mat mDepth8UC1 = new Mat(mDepth.rows(), mDepth.cols(), CV_8UC3);
                                                            mDepth.convertTo(mDepth8UC1, CV_8UC3, 1 / 255.0);
                                                            Log.e(TAG, "streaming: before  saving bitmap4 ");

                                                            Bitmap bmpDisplay = Bitmap.createBitmap(mDepth8UC1.cols(), mDepth8UC1.rows(), Bitmap.Config.ARGB_8888);
                                                            Utils.matToBitmap(mDepth8UC1, bmpDisplay);
                                                            String path = saveToInternalStorage(bmpDisplay);
                                                            Log.e(TAG, "streaming: before  saving bitmap5 " + path);

                                                        }
                                                    } catch (Exception ex) {
                                                        Log.e(TAG, "streaming: after saving bitmap " + ex.getMessage());
                                                    }

                                                }

                                            });

                                            mGLSurfaceView.upload(processed);
                                        }
                                    }
                                }
                            }
                            pipeline.stop();
                        } catch (Exception e) {
                            Log.e(TAG, "streaming, error: " + e.getMessage());
                        }
                    }
                }
            }
        }
    };

MartyG-RealSense, 2019-12-26:
I am wondering if you need to set the color scheme with **RS2_OPTION_COLOR_SCHEME**

https://github.com/IntelRealSense/librealsense/issues/5287

kafan1986, 2019-12-27:
You can achieve it in either of two ways:
A) Use colorizer filter provided by the librealsense SDK to convert the depth data to RGB scheme and then this is just like RGB frame and save it.
B) Apply colormap to the 8UC1 image (gray) of your depth frame.

RealSenseCustomerSupport, 2020-01-07:
----------------------------------------------

Please let us know if more help is needed.
Thank you.

closed 2020-01-10 by RealSenseCustomerSupport