realsenseai/librealsense #5546: try to save 16UC1 depth frame into .png

yanxh97, 2019-12-27:
Using D435 on Ubuntu 16.04.  C++.  OpenCV3.1.0
I've tried merge save-to-disk.cpp and align.cpp to create a RGBD data set like TUM, which consists of 8UC3 RGB images and 16UC1 depth images. However, the examples always colorize the depth image into a 8UC3 image and save/display it. It seems that all of the 10 schemes are doing so.

Good news is that we have the function depth_frame::get_data(). The format of the data is RS2_FORMAT_Z16. Is it the same as 16UC3 in OpenCV?

Can you provide me with an example of how to load raw depth data into a cv::Mat and save it as a png? Thanks a lot. (stbi_write_png do not support 16-bit png and that's why I turn to OpenCV)

yanxh97, 2019-12-27:
I've solved the problem to some extent by:

`cv::Mat depth_image(cv::Size(width,height) , CV_16UC1 , (void *)depth_frame.get_data() , cv::Mat::AUTO_STEP)`

But new problems occur. The images are so dark compared to those of TUM's. It may be the scale problem. TUM's Kinect2 has a 5000 scale while rs2 D435's is only 1000. But when I use imwrite("name.png", depth_image*5), the situation does not change much until the factor are set to about 15.

ev-mp, 2020-01-06:
@yanxh97 hello, note that [PNG assumes big-endian order](https://stackoverflow.com/questions/2384111/png-file-format-endianness) while the [Depth Z16 data is little-endian](https://linuxtv.org/downloads/v4l-dvb-apis/uapi/v4l/pixfmt-z16.html). 
So in order to use `frame.get_data()` with PNG you actually have to introduce an intermediate processing step to reverse the order of all bytes, which, imho, renders the PNG impractical w.r.t. Realsense Depth format.
Duplicate of #2147

yanxh97, 2020-01-07:
@ev-mp Thanks for your reply. It seems that I'm kind of lucky because the constructor of cv::Mat also assumes little-endian order, the same as `frame.get_data()`. The real problem, however, is that half of my image1 is totally black, while the other part is correct. So I guess it is the cv::Mat & the memory problem.

closed 2020-01-07 by yanxh97