realsenseai/librealsense #5547: Problem with D435i imu calibration

lishanggui, 2019-12-27:
Required 				Info |  
-- | --
Camera 				Model | D435i
Firmware 				Version | 05.12.01.00
Operating 				System & Version | Ubuntu 				16.04
Kernel 				Version (Linux Only) | (e.g. 				4.15.0)
Platform | PC
SDK 				Version | 2.30.0
Language | C++

I used D435i sensor box as recommended in "" librealsense / tools / rs-imu-calibration / images / "to ensure that the posture's horizontal / vertical alignment is tested. The D435i was calibrated multiple times with the librealsense tool rs_imu_calibration.py. (I repeated all experiments in # 5177, and it has the same problem)

One calibration result is as follows:

Direction data collected.
Would you like to save the raw data? Enter footer for saving files (accel_<footer>.txt and gyro_<footer>.txt)
Enter nothing to not save raw data to disk. >

```

Not writing to files.
[-0.00122304 -0.00134677 -0.0019687 ]
[1000 1000 1000 1000 1000 1000]
using 6000 measurements.
[[ 1.01675963  0.01071049 -0.03569976]
 [-0.01154162  1.0245261   0.02870147]
 [ 0.01067121 -0.04971456  1.00277935]
 [ 0.09659864 -0.01756813  0.14565727]]
residuals: [  21.59537199  108.52902841  234.33421158]
rank: 4
singular: [ 440.32710438  427.97283761  424.08900959   77.42098444]
norm (raw data  ): 9.630912
norm (fixed data): 9.776162 A good calibration will be near 9.780300
Would you like to write the results to the camera's eeprom? (Y/N)y
Writing calibration to device.
  Device PID:  0B3A
  Device name:  Intel RealSense D435I
  Serial number:  912112071569
  Firmware version:  05.12.01.00
SUCCESS: saved calibration to camera.
Done.
```

However, right after this when I check the acc. Output in realsense-viewer, it is not close to 9.7803 (the local acceleration of gravity is 9.7803).

I tested with realsense-viewer, The orientation of the device is set to 6 different cases according to the calibration[ white paper](https://www.intel.com/content/www/us/en/support/articles/000032303/emerging-technologies/intel-realsense-technology.html). (Section 4.3.2)

Position 1:   9.759(motion correction on)    9.535(motion correction off)
Position 2:  10.009(motion correction on)    9.756(motion correction off)
Position 3:   9.810(motion correction on)    9.556(motion correction off)
Position 4:   9.581(motion correction on)    9.524(motion correction off)
Position 5:   9.230(motion correction on)    9.350(motion correction off)
Position 6:   10.338(motion correction on)   10.150(motion correction off)

In rs-motion.cpp main function, I added a couple of lines to check the value in the terminal as below.
```
// Get accelerometer measures
rs2_vector accel_data = motion.get_motion_data ();
```
the result:
Position 1: 9.7678
Position 2: 10.0618
Position 3: 9.82553
Position 4: 9.58106
Position 5: 9.26782
Position 6: 10.3381

Run 'rs-enumerate-devices -c' and the result is as follows
Motion Intrinsic Parameters:

Motion Intrinsic of "Gyro" MOTION_XYZ32F
Bias Variances: 0.000000000000000 0.000000000000000 0.000000000000000
Noise Variances: 0.000000000000000 0.000000000000000 0.000000000000000
Sensitivity:
     1.000000 0.000000 0.000000 -0.000021
     0.000000 1.000000 0.000000 -0.000024
     0.000000 0.000000 1.000000 -0.000034

Motion Intrinsic of "Accel" MOTION_XYZ32F
Bias Variances: 0.000000000000000 0.000000000000000 0.000000000000000
Noise Variances: 0.000000000000000 0.000000000000000 0.000000000000000
Sensitivity:
     1.016760 -0.011542 0.010671 0.096599
     0.010710 1.024526 -0.049715 -0.017568
    -0.035700 0.028701 1.002779 **0.145657**

The accelerometer's sensitivity matrix shows a notable bias for the third axis compared to the first two.

How can the problem be solved? Did the engineer reproduce the problem and get it resolved? I am currently using realsense D435i to do slam experiments, and I am more anxious to solve this problem.

patbenatar, 2019-12-27:
(Perhaps related: https://github.com/IntelRealSense/librealsense/issues/5177)

I'm experiencing what sounds like the same issue.

After multiple attempts at `python rs-imu-calibration.py` using a rigid frame built and verified with a level, I'm getting accelerations notably deviating from gravity in certain orientations as well as significant values on the axes that I would expect to be near-zero.

## Calibration

```
Motion Intrinsic Parameters:

Motion Intrinsic of "Gyro"	  MOTION_XYZ32F
Bias Variances: 	0.000000000000000  0.000000000000000  0.000000000000000
Noise Variances: 	0.000000000000000  0.000000000000000  0.000000000000000
Sensitivity :
     1.000000       0.000000       0.000000       0.000001
     0.000000       1.000000       0.000000       0.000008
     0.000000       0.000000       1.000000      -0.000041

Motion Intrinsic of "Accel"	  MOTION_XYZ32F
Bias Variances: 	0.000000000000000  0.000000000000000  0.000000000000000
Noise Variances: 	0.000000000000000  0.000000000000000  0.000000000000000
Sensitivity :
     1.018967      -0.004261       0.000489      -0.012745
     0.034915       1.027532      -0.005286       0.206483
    -0.034412      -0.004820       1.010221       0.472612
```

## Observed accelerations

* Position 1
  ```
  x: -0.0824612751603
  y: -9.84611606598
  z: 0.943550467491
  ```
* Position 2
  ```
  x: 9.77985572815
  y: -0.669295370579
  z: 0.854578495026
  ```
* Position 3
  ```
  x: 0.0589955076575
  y: 9.81116771698
  z: 0.994111776352
  ```
* Position 4
  ```
  x: -9.82306098938
  y: 0.695441424847
  z: 1.07413542271
  ```
* Position 5
  ```
  x: -0.00596596579999
  y: 0.0567697882652
  z: -8.84863853455
  ```
* Position 6
  ```
  x: -0.0373049341142
  y: -0.0409596562386
  z: 10.7277708054
  ```

Any help for those of us struggling with D435i calibration would be much appreciated!

lishanggui, 2020-01-06:
The rs_imu_calibration.py provided by Intel realsense requires that each position ([write paper](https://www.intel.com/content/www/us/en/support/articles/000032303/emerging-technologies/intel-realsense-technology.html)) needs to be completely horizontal or vertical during the calibration process. It is difficult to achieve absolute horizontal or vertical experimental conditions, and the accuracy of the calibration will be affected. Recently, I used other methods for calibration, and wrote the calibration results to the device through the rs_imu_calibration.py writer. I found that the results output using realsense-viewer and rs-motion are the same as above, and I suspected that it was an input device There was a problem with the post-correction imu data conversion process. Then I did this kind of experiment, let the equipment output the imu raw data, and use the calibration data to transform it by a program written by myself. As a result, the equipment was placed in a different way and the gravity acceleration was 9.8. Therefore, it is concluded from these experiments that the main reason for the inaccurate results after imu calibration is the problem of the underlying conversion. At the same time, I think that rs_imu_calibration.py is not a particularly good method, because reducing the accuracy requires more experimental equipment Horizontal and vertical.

ev-mp, 2020-01-06:
@lishanggui hello,
Thank you for the insights.
Obviously the calibration precision is limited as it lacks ground truth and shifts this responsibility to the user. Thus the main purpose of `rs_imu_calibration.py` is to minimize the errors by adjusting both for the intrinsic and extrinsic sensor matrix.

And even in case of external equipment used to verify horizontal/vertical alignment there would still remain fundamental questions unresolved, such as to what degree the acceleration vector, lets say, is skewed from the plumb line due to its mechanical posture (mechanical) rather than the sensor's analog/digital bias. (The equipment can physically measure the orientation of the camera's case but not the IMU sensor)

Additional comments:
>I found that the results output using realsense-viewer and rs-motion are the same as above, and I suspected that it was an input device There was a problem with the post-correction imu data conversion process

Note that writing the IMU calibration to the flash is only the first step. and to use the motion correction in streaming requires to check and turn on the `RS2_OPTION_ENABLE_MOTION_CORRECTION` option.

lishanggui, 2020-01-11:
When I did the above experiments, I confirmed that RS2_OPTION_ENABLE_MOTION_CORRECTION was turned on when running realsense_viewer and rs_motion. Only when I converted by my own program, I turned off RS2_OPTION_ENABLE_MOTION_CORRECTION and then converted; whether you have repeated these experiments, and the results you obtained are What is it？

lishanggui, 2020-01-11:
@ev-mp When I did the above experiments, I confirmed that RS2_OPTION_ENABLE_MOTION_CORRECTION was turned on when running realsense_viewer and rs_motion. Only when I converted by my own program, I turned off RS2_OPTION_ENABLE_MOTION_CORRECTION and then converted.   whether you have repeated these experiments, and the results you obtained are What is it？

In the previous stages, I have seen the content of your imu calibration program and the content of rs_motion program.

ev-mp, 2020-05-04:
@lishanggui, what is the current status on this?
Please update

RealSenseSupport, 2020-05-15:
Hi @lishanggui 

The latest update to D435i IMU calibration came in this PR https://github.com/IntelRealSense/librealsense/pull/6029

Is there any status on this?

*If we don’t hear from you in 7 days, this issue will be closed.

lishanggui, 2020-05-21:
@ev-mp @RealSenseSupport     In the new version, this problem has been resolved. Thank you！

closed 2020-05-21 by lishanggui