realsenseai/librealsense #5549: Not getting single image from frame

lokeshgs87, 2019-12-27:
void GenerateColorImage(Frame frame)
    {
        VideoFrame videoFrame = frame.as(Extension.VIDEO_FRAME);

        Log.e(TAG, "streaming:  " + videoFrame.getHeight() + " " + videoFrame.getWidth() + " " + videoFrame.getBitsPerPixel() );

        Log.e(TAG, "streaming: before  saving bitmap1 ");

        Mat mDepth = new Mat(videoFrame.getHeight(), videoFrame.getWidth(), CV_16UC3);

//        Mat image8u3 = new Mat();
//        mDepth.convertTo(image8u3, CV_8UC3); // CV_8U should work as well
//        mDepth = image8u3;

        Log.e(TAG, "streaming: before  saving bitmap2 ");

        int size = (int) (mDepth.total() * mDepth.elemSize());
        byte[] return_buff = new byte[size];
        //  depthFrame.getData(return_buff)

        //  byte[] depth_buff = new byte[depthFrame.getHeight() * depthFrame.getStride()];
        videoFrame.getData(return_buff);
        short[] shorts = new short[size / 2];
        ByteBuffer.wrap(return_buff).order(ByteOrder.nativeOrder()).asShortBuffer().get(shorts);

        Log.e(TAG, "streaming: before  saving bitmap2.1 ");
        mDepth.put(0, 0, shorts);
        Log.e(TAG, "streaming: before  saving bitmap3 ");

        Mat mDepth8UC1 = new Mat(mDepth.cols(), mDepth.rows(), CV_8UC4);
        mDepth.convertTo(mDepth8UC1, CV_8UC4, 1 / 255.0);
        Log.e(TAG, "streaming:  " + mDepth8UC1.cols() + " " +  mDepth8UC1.rows() + " " + Bitmap.Config.ARGB_8888 );

        Bitmap bmpDisplay = Bitmap.createBitmap(mDepth8UC1.cols(), mDepth8UC1.rows(), Bitmap.Config.ARGB_8888);
        Utils.matToBitmap(mDepth8UC1, bmpDisplay);
        String path = saveToInternalStorage(bmpDisplay,"color.png");
        Log.e(TAG, "streaming: before  saving bitmap5 " + path);

    }
[image]

lokeshgs87, 2019-12-27:
please find above image which we are getting. We wanted single image

RealSenseSupport, 2020-07-25:
Hi,

Will you still be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

closed 2020-08-14 by RealSenseSupport