realsenseai/librealsense #5551: Where is A-Factor (via Disparity Modulation)

goatShaver, 2019-12-27:
C/C++, librealsense 2.31.0 (I am running FW 5.12+) on 435 and 435i.

I can find no reference or code with A-Factor per the below document. I would like to be able to control it. (Am already using advanced mode, just can't see where to set A-Factor in current 2.31.0 release.)

https://dev.intelrealsense.com/docs/white-paper-subpixel-linearity-improvement-for-intel-realsense-depth-cameras

MartyG-RealSense, 2019-12-28:
I was only able to find Python code for getting and setting the A-Factor.  I hope that it will help you to work out a C++ version.  Someone on this forum may also be able to offer a script.

https://github.com/IntelRealSense/librealsense/issues/5277

goatShaver, 2019-12-28:
Thanks for the lead, for C++, the call is <s|g>et_amp_factor(). The documentation refers to it as A-Factor, however the SDK refers to it as amplitude.

closed 2019-12-28 by goatShaver