realsenseai/librealsense #5553: Measure the angle between the camera and an object

maky-hnou, 2019-12-28:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D435  | 
| Firmware Version                   | (05.12.01.00) | 
| Operating System & Version |  Ubuntu 18.04  | 
| Kernel Version (Linux Only)    |  5.0.0-37-generic                                         | 
| Platform                                 | PC |
| SDK Version                            |  pyrealsense 2.30.0.1216  | 
| Language                            |  C++/Python  | 
| Segment			|  Robot  | 

### Issue Description
I recently started working with an Intel depth camera D435 and started discovering the SDK functionalities.  
I know it is possible, using the built-in functions, to get the (x, y, z) coordinates of a pixel (or an object) and measure the distance between the camera and that pixel (or that object).  
I wonder if there is a built-in function that measures the (inclination) angle between the camera and an object.  (Check the image below)

[image]  
Say that the object and the camera exist in the same plane (which is defined in the image above by **(the camera, Y, Z)**), is there a built in function that measures the angle between the object and the Y axis? If not, is there any idea how to do that based on the information ((x, y, z) and distance) we get from the depth camera?

kafan1986, 2019-12-28:
@maky-hnou You can consider any point as your reference center. Let's  assume center of the image is your camera center then as the camera characteristics for resolution is prior known (look at the D400 specs). You can easily get the horizontal angle and vertical angle for any pixel by dividing it the pixel distance divided by length or width of image multiplied by corresponding horizontal FOV or Vertical FOV.

maky-hnou, 2019-12-28:
@kafan1986 , as I understood:  

1. get the distance between the camera and the pixel using: `float distance = get_distance(x, y);  // x and y are pixel coordinates`.  

2. divide the obtained distance by the width of the image (except that I didn't understand what do you mean by the width of the image, is it the width in pixels or the maximum distance of the FOV? If so, how to to calculate it?): `float distance_by_width = distance / image_width; // image_width?` 

3. get the vertical angle: `float angle = distance_by_width * vertical_depth_FOV; //vertical_depth_FOV = 62 for D435.`   

Is that what you meant? And would you please clarify what do you mean by " length or width of image"?

kafan1986, 2019-12-29:
@maky-hnou 
Why would you need the distance to find  the angle?

Example:
Say you are using RGB camera and have set it to 720p resolution. W = 1280, H = 720
Assume the HFOV = 62 and VFOV=46 (Check values for your camera, assumed at random)
Center of camera in image = (W/2,H/2) = (640,360)

Horizontal Angle of say random pixel (x,y) = ((x - W/2)/(W/2))*(HFOV/2)
Vertical Angle of say random pixel (x,y) = ((y - H/2)/(H/2))*(VFOV/2)

For a random pixel say (320,180) it will be:
H Angle = ((320 - 640)/640)*(62/2) = - 15.5 degree
H Angle = ((180 - 360)/360)*(46/2) = - 11.5 degree

You can get the euclidean angle by using simple (h-angle^2+v-angle^2)^0.5

maky-hnou, 2019-12-29:
@kafan1986  that would be right if I wanted to know the angle of a pixel according to the image center or according to any other other pixel of the image, but I want to know the angle between a pixel of the image and the camera that took that image.  
I guess this approach does not solve my problem..

ev-mp, 2020-01-06:
@maky-hnou hello, the description is somewhat confusing. @kafan1986 did try to outline the calculation for the `alpha` denoted in the above sketch.

Please correct me if I'm wrong but it seem that what you're actually looking for is the angle between the **normal of the pixel on target**  (which doesn't appear in the sketch) and the camera's line of site. 
And if that's correct then finding the inclination can be reduced to finding the target's plane or its normal, which are actually the same. We do plane fit in the SDK [here](https://github.com/IntelRealSense/librealsense/blob/d39b5139ce62efffddee4cd11a87e8927ecdfa0c/tools/depth-quality/depth-metrics.h#L62-L63) to calculate depth quality metrics, which allows to find the object's plane in certain conditions
```c++
    ....
        //Based on: http://www.ilikebigbits.com/blog/2015/3/2/plane-from-points
        inline plane plane_from_points(const std::vector<rs2::float3> points)
```

maky-hnou, 2020-01-07:
@ev-mp that's exactly what I am looking for. Thank you so much!

closed 2020-01-07 by maky-hnou

jianjuan, 2021-08-23:
> @ev-mp that's exactly what I am looking for. Thank you so much!

Can I make a conclusion that you are looking for pose between the object and the camera? The centroid of 3D points as (x, y, z) and the normal vector as (u, v, w) or (rx, ry, rz)?
Looking forward to your reply, thanks!

pfcouto, 2022-09-20:
> @maky-hnou Why would you need the distance to find the angle?
>
> Example: Say you are using RGB camera and have set it to 720p resolution. W = 1280, H = 720 Assume the HFOV = 62 and VFOV=46 (Check [...]

HI, @maky-hnou, I am trying to calculate the same thing as you. The D435 camera as diferent sensors with different FOVs, therefore the confusion. Can you tell me which FOV you used and why you used that one and not one of the others? 

[image]

maky-hnou, 2022-09-20:
@pfcouto it has been a long time since I worked on that project.
Unfortunately, I don't remember any details about it.

pfcouto, 2022-09-20:
@maky-hnou No problem! Thanks!

kay2019-art, 2024-10-24:
I understand the scheme you said, then how to get the Angle? Is there any relevant code case @ev-mp

MartyG-RealSense, 2024-10-24:
Hi @kay2019-art  The links below have some further reading about getting the angle that may be helpful.

https://github.com/IntelRealSense/librealsense/issues/10043

https://answers.ros.org/question/300639/how-to-get-the-distance-and-the-angle-distance-and-bearing-from-a-depth-camera/

kay2019-art, 2024-11-02:
Hello, this is the code I wrote according to your idea to calculate the horizontal and vertical Angle between the camera and the detected object. The idea is as follows: 1. Import yolo model to realize the coordinate position of the object detection on the upper left and lower right; 2. Map the two-dimensional coordinate points to the three-dimensional coordinate points, 3. Take four points to form a plane, get the plane equation, 4. The normal vector is obtained by plane equation to calculate the Angle between horizontal and vertical. Please check whether this is reasonable. The result obtained is not the desired part due to errors in the code. Please also ask engineers to modify my code, thank you！

import cv2
import pyrealsense2 as rs
import numpy as np
from ultralytics import YOLO

class Plane:
    def __init__(self, a, b, c, d):
        self.a = a  # Coefficient of x in the plane equation
        self.b = b  # Coefficient of y in the plane equation
        self.c = c  # Coefficient of z in the plane equation
        self.d = d  # D coefficient in the plane equation

def plane_from_point_and_normal(point, normal):
    d = -np.dot(normal, point)  # Calculate D coefficient
    return Plane(normal[0], normal[1], normal[2], d)

def plane_from_points(points):
    if len(points) < 3:
        raise ValueError("Insufficient points to calculate a plane")
    
    centroid = np.mean(points, axis=0)  # Calculate the centroid of the points
    cov_matrix = np.zeros((3, 3))  # Initialize covariance matrix
    for point in points:
        temp = point - centroid
        cov_matrix += np.outer(temp, temp)  # Build covariance matrix

    eigvals, eigvecs = np.linalg.eig(cov_matrix)  # Calculate eigenvalues and eigenvectors
    normal = eigvecs[:, np.argmin(eigvals)]  # Get the normal vector
    return plane_from_point_and_normal(centroid, normal)

def calculate_angles_plane(plane):
    normal = np.array([plane.a, plane.b, plane.c])
    horizontal_angle = np.arctan2(normal[1], normal[0]) * (180 / np.pi)  # Calculate horizontal angle
    vertical_angle = np.arctan2(normal[2], np.linalg.norm(normal[:2])) * (180 / np.pi)  # Calculate vertical angle
    return horizontal_angle, vertical_angle

class DepthCamera:
    def __init__(self):
        self.pipeline = rs.pipeline()  # Initialize the RealSense pipeline
        self.config = rs.config()
        self.config.enable_stream(rs.stream.depth, 640, 480, rs.format.z16, 30)  # Configure depth stream
        self.pipeline.start(self.config)  # Start the pipeline
        self.depth_frame = None

    def update_frames(self):
        frames = self.pipeline.wait_for_frames()  # Wait for a new set of frames
        self.depth_frame = frames.get_depth_frame()  # Get the depth frame

    def get_depth_at_point(self, x, y):
        if self.depth_frame:
            width, height = self.depth_frame.get_width(), self.depth_frame.get_height()
            if 0 <= x < width and 0 <= y < height:
                return self.depth_frame.get_distance(x, y)  # Return depth at the specified pixel
        return None

    def get_world_coordinates(self, x, y):
        depth_value = self.get_depth_at_point(x, y)  # Get depth at (x, y)
        if depth_value is None:
            return None
        depth_intrin = self.depth_frame.profile.as_video_stream_profile().intrinsics
        return rs.rs2_deproject_pixel_to_point(depth_intrin, [x, y], depth_value)  # Convert to world coordinates

def calculate_actual_size_and_angles(camera, x1, y1, x2, y2):
    points = [
        camera.get_world_coordinates(x1, y1),
        camera.get_world_coordinates(x2, y2),
        camera.get_world_coordinates(x2, y1),
        camera.get_world_coordinates(x1, y2),
    ]
    
    points = [p for p in points if p is not None]  # Filter out None values
    
    if len(points) < 3:
        raise ValueError("Not enough valid 3D points to fit a plane")

    best_fit_plane = plane_from_points(points)  # Fit a plane to the points
    print(f"Plane equation: {best_fit_plane.a}x + {best_fit_plane.b}y + {best_fit_plane.c}z + {best_fit_plane.d} = 0")
    
    horizontal_angle, vertical_angle = calculate_angles_plane(best_fit_plane)  # Calculate angles
    print(f"Horizontal angle: {horizontal_angle:.2f}°")
    print(f"Vertical angle: {vertical_angle:.2f}°")
    
    return horizontal_angle, vertical_angle

if __name__ == "__main__":
    # Import YOLO model for object detection
    img = cv2.imread("test.jpg")
    model = YOLO('model/best.pt')
    results = model(img)

    # Extract bounding box coordinates after detection
    dec_xyxy = results[0].boxes.xyxy
    if dec_xyxy.size > 0:
        xmin, ymin, xmax, ymax = map(int, dec_xyxy[0][:4])
        
        camera = DepthCamera()  # Create a DepthCamera instance
        camera.update_frames()  # Update the frames to get the latest depth data
        horizontal_angle, vertical_angle = calculate_actual_size_and_angles(camera, xmin, ymin, xmax, ymax)
    else:
        print("No objects detected.")

kay2019-art, 2024-11-02:
@ev-mp @MartyG-RealSense Please check whether there is any problem with the code and whether it can be modified into a correct result, so that it can be completely used for your reference, thank you!

MartyG-RealSense, 2024-11-02:
@kay2019-art  I do not have sufficient knowledge of YOLO programming or the described calculations to offer advice about the correctness of your code unfortunately and are unable to refer scripts to Intel RealSense colleagues for debugging to produce a complete working script.  I do apologize.

I would recommend using 848x480 depth resolution instead of 640x480 if possible to do so, as accuracy will reduce if the resolution used is lower than 848x480 (which is the optimal depth accuracy resolution for RealSense 400 Series cameras).

kay2019-art, 2024-11-02:
It doesn't matter if you don't understand YOLO, you can set these two points as fixed, mainly to verify whether the following process and calculation Angle are correct. I don't have any good suggestions

MartyG-RealSense, 2024-11-02:
I am not able to run and test scripts on my computer, only read the code pasted in comments and make suggestions.  Most of the calculations used in this particular script are outside of my programming knowledge though.

Can you describe what aspect of the output result of the script is undesirable, please?  Thanks very much for your patience.

kay2019-art, 2024-11-03:
By calculating the Angle between the normal vector and the plane to calculate the Angle between the horizontal and vertical, the ideal output Angle is, I move to the left for a positive Angle, the right for a negative Angle, through the camera center is zero Angle, but actually the Angle change is positive, I would like to ask if there is any problem in the post-processing Angle calculation?

MartyG-RealSense, 2024-11-03:
In the coordinate system of RealSense cameras, positive horizontal movement is in the right direction rather than left.  

https://github.com/IntelRealSense/librealsense/blob/master/doc/d435i.md#sensor-origin-and-coordinate-system