realsenseai/librealsense #5554: T265: Weird problem with a valid code sequence

neilyoung, 2019-12-28:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.896 |
| Operating System & Version |  macOS Catalina |
| Kernel Version (Linux Only)    |      | 
| Platform                                 |   |
| SDK Version                            |   2.31.0, development, #5213 applied                         | 
| Language                            |                     | 
| Segment			|  others                           | 

### Issue Description

I'm sorry, but I'm having again some weird but reproducible problem with the T265. 

I have tried to break down the problem to a simple script. The frame callback parts are mostly taken from the `t265_stero.py` example. Please bear with me for the mix of class variables and local variables. The code is taken as is from a larger script.

The problem: With a particular, but valid line contained, the _second_ start fails with 

```
ERROR:root:Exception in pose_handler
Traceback (most recent call last):
  File "pose-test.py", line 52, in start
    device = config.resolve(self.__pipeline).get_device()
RuntimeError: No device connected

```

I have no clue, why. The line in question is commented with 

 `# This line causes an exception "No device connected" at second start`

When the script starts it waits for ENTER to start. After start wait a second until the first pose trace appears. Then press ENTER again to stop. With the special line in place the second start should fail. If this line is commented it works fine.

Here is the script:

```
#!/usr/bin/python
# -*- coding: utf-8 -*-

import pyrealsense2 as rs
import threading
from threading import Lock
import logging

class Pose:
    def __init__(self):
        self.__pose_sensor = None
        self.__pose_handler_thread = None
        self.__pipeline = None
        self.__frame_mutex = Lock()
        self.__frame_data = {"pose": None, "timestamp": None}
        self.__stopped = False
        self.__last_reported_raw_frame_number = 0
   

    # T265 frame callback
    def __frame_callback(self, frame):
        """This callback is called on a separate thread, so we must use a mutex to ensure that data is synchronized properly. We should also be
           careful not to do much work on this thread to avoid data backing up in the callback queue.
        """
        pose_frame = frame.as_pose_frame()
        self.__frame_mutex.acquire()
        self.__frame_data["pose"] = pose_frame
        self.__frame_data["timestamp"] = pose_frame.timestamp
        self.__frame_mutex.release()

    # Stop operation
    def stop(self):
        '''Stop operation'''
        self.__stopped = True
        self.__pose_handler_thread.join()
        self.__pipeline.stop()

    # Start operation
    def start(self):
        '''Start operation'''
        try:
            self.__pipeline = rs.pipeline()
            config = rs.config()
            config.enable_stream(rs.stream.pose)
            device = config.resolve(self.__pipeline).get_device()

            # This line causes an exception "No device connected" at second start
            self.__pose_sensor = device.first_pose_sensor()
 
            self.__pose_handler_thread = threading.Thread(target=self.__pose_handler_loop)
            self.__pose_handler_thread.daemon = True
            self.__pose_handler_thread.start()
            self.__pipeline.start(config, self.__frame_callback)

        except Exception:
            logging.exception("Exception in pose_handler")

    def __pose_handler_loop(self):
        while not self.__stopped:
            try:
                self.__frame_mutex.acquire()
                valid = self.__frame_data["timestamp"] is not None
                self.__frame_mutex.release()

                if valid:
                    self.__frame_mutex.acquire()
                    pose = self.__frame_data["pose"]
                    self.__frame_data["timestamp"] = None
                    self.__frame_mutex.release()
                    
                    data = pose.get_pose_data()

                    if pose.frame_number - self.__last_reported_raw_frame_number >= 200:
                        self.__last_reported_raw_frame_number = pose.frame_number
                        print("raw pose: confidence: {}, translation: {}, rotation: {}".format(data.tracker_confidence, data.translation, data.rotation))

            except Exception:
                logging.exception("Exception in pose_handler")

while True:
    input("Press Enter to START...\n")
    pose = Pose()
    pose.start()
    input("Press Enter to STOP...\n")
    pose.stop()

```

neilyoung, 2019-12-30:
Could anybody please confirm this issue?

neilyoung, 2019-12-30:
Update: The problem is there, even if I upgrade to fw 0.2.0.908. No change. Second start fails.

neilyoung, 2020-01-02:
I have tested with the latest development branch now, and - as expected - this is still a problem.

It is not possible to start/stop/start a T265 if you mess with a `pose_sensor`

neilyoung, 2020-01-03:
@bfulkers-i Hi Brian, sorry for contacting you directly, but would it be possible for you to keep an eye on this problem? Thanks in advance.

neilyoung, 2020-01-06:
Dear community, I would really appreciate, if an owner of a T265 could simply try the python script above in order to confirm (or not) the issue. I not even expect it to get fixed anymore.

Working around by restarting my script after each session now. Horrible...

radfordi, 2020-01-07:
Hi @neilyoung, I was able to confirm that something is keeping the device open after `pose` is destroyed` even if I change your loop to:
```python
while True:
    input("Press Enter to START...\n")
    pose = Pose(ctx)
    pose.start()
    input("Press Enter to STOP...\n")
    pose.stop()
    pose = None
    gc.collect()
```
I'll try and figure out what's going on, but @dorodnic might be able to see what's going on here quicker than I can.

neilyoung, 2020-01-07:
@radfordi Ah, now... Thank you very much for the confirmation. Makes my day :)

radfordi, 2020-01-07:
@neilyoung, this patch "fixes" it for me, but I don't understand the reference counting enough yet to know if there is an actual bug yet.   @dorodnic?
```patch
         self.__stopped = True
         self.__pose_handler_thread.join()
         self.__pipeline.stop()
+        self.__pose_sensor = None

     # Start operation
```

neilyoung, 2020-01-07:
Wow. That simple. Should have seen that too.. Embarrassing...

Anyway, IMHO it shouldn't be necessary to do that, but indeed, it "fixes" it.

neilyoung, 2020-01-15:
I think it should not be closed. There is a “solution” but not a fix

radfordi, 2020-01-15:
I agree @neilyoung, this is not fixed and is a regression.

neilyoung, 2020-01-31:
closed for lack of interest

closed 2020-01-31 by neilyoung