realsenseai/librealsense #5555: Is navigation feasible using D435i & T265 together,  without wheel encoders.

antoan, 2019-12-28:
I am working with a 4 wheeled skid-steered rover which has no wheel encoders available.

The T265 section of the ros/realsense front page has the following notice: 
"Important Notice: For wheeled robots, odometer input is a requirement for robust and accurate tracking." 
https://github.com/IntelRealSense/realsense-ros#using-t265

Is the above still relevant in the case of using D435i & T265 together, rather than a single T265 camera? 

I am trying determine if I can avoid the installation of wheel encoders which may be very difficult to setup on my platform.

neilyoung, 2019-12-28:
It is really hard to give a definite answer for that, since it depends on your requirements. From my experiences it is surely possible to achieve "pedestrian navigation" out of the box with a certainty of +-50cm indoors in an  area of 100 x 100 m. But even this depends on a lot of parameters: Number of features, lighting conditions, quality of mapping. But I would give it a try. I'm not even using a D435 as companion device; I'm just playing around with the T265 and I'm not too disappointed for now. I would encourage you to try by yourself with a stand-alone T265 first in order to see, if it meets your requirements.

radfordi, 2020-01-08:
Just to add a bit to what @neilyoung said.   If your platform doesn't have a lot of vibrations and can avoid passing jolts from the ground to the T265, then you might be able to get away without wheel encoders.  The setup is significantly easier without them.

antoan, 2020-01-17:
@neilyoung radfordi
Thank you for your input, it is just what I was looking for. Much appreciated.

closed 2020-01-17 by antoan