realsenseai/librealsense #5556: T265: A save way to provoke an exception while attempting to save a map

neilyoung, 2019-12-29:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   | 0.2.0.896 |
| Operating System & Version |  macOS Catalina |
| Kernel Version (Linux Only)    |      | 
| Platform                                 |   |
| SDK Version                            |   2.31.0, development, #5213 applied                         | 
| Language                            |                     | 
| Segment			|  others                           | 

### Issue Description

It is pretty simple:  Try to save a map early after brand new start. Happens regardless of pose tracking confidence. The reproducible exception is

`null pointer passed for argument "buffer"
`

I think there could be a better way to express the inability to save a map, e.g. if there are not enough map points internally. A None for output_map would do it :)

The exception is the answer to this call:

`            output_map = self.__pose_sensor.export_localization_map()
`

neilyoung, 2019-12-30:
Problem seems to be fixed with fw 0.2.0.908, #5504

neilyoung, 2020-01-08:
Nope, it is still there :) Most likely an easy to fix thingy...

neilyoung, 2020-01-20:
Any update here? `null pointer passed for argument "buffer"` is the common answer to nearly 50 percents of all attempts to store a map.

RealSenseSupport, 2020-10-12:
Hi @neilyoung 

Are you still experiencing this on the latest T265 f/w (v. 0.2.0.951)?

Thanks

RealSenseSupport, 2020-10-12:
Nvm...was able to reproduce by forcing an error on latest, so disregard last question.

That said,...Thanks for highlighting this. I will pass this enhancement request on to our dev team.

Thanks again.

neilyoung, 2020-10-13:
@RealSenseSupport Hmm. Tried right now with 2.39.0. Wasn't able to reproduce so far. Firmware is 0.2.0.951

RealSenseSupport, 2020-10-26:
Hi @neilyoung 

That's good to hear. The way I forced the error would never occur in practice. :)  

And it has been a while so hopefully this has been improved since. 

That said, at this time we _have_ moved our focus to our next generation of products and consequentially will not be addressing new T265 related enhancements.

Thanks

closed 2020-11-04 by RealSenseSupport