realsenseai/librealsense #5557: rs-convert tool problem

Shivani1796, 2019-12-30:
Hello 
I want to convert my .bag file to CSV file which i recorded by " t265 " Camera for getting the translation and rotation matrix  for this i am using the converter tool but when i try this it works till 99% and prints CSV Conveter and no output get generated , others like bag to png and all are working properly but this bag to CSV does not work for me.
Is their any solution for this? 
Thanks ..

RealSenseCustomerSupport, 2020-01-06:
----------------------------------------------

Hi @Shivani1796

Can you help me with your setup and environment?  The bag file that you generated, was it done via the RealSense Viewer or via some other method?  If some other method please outline what method was used?  Are you using the latest librealsense 2.31.0?  What streams are enabled (pose only or a combination of IMU data, pose and/or camera streams)?

Shivani1796, 2020-01-07:
@RealSenseCustomerSupport Hey, thanks for the reply,
I am using realsense T265 camera , on linux OS . Yes using the latest librealsense 2.31.0
Bag file is created by realsense-viewer. 

[image]

The above picture is , I played my bag file on realsense-viewer here I can see pose,accel,gyro and want to retrieve the same data in a text file/csv file . When I run rs-convert tool it don't work properly and gives error Segmenation error (core dumped).
**At the end I want to retrieve all data like gyro, accel, pose (rotation &  translation matrix)  for all the frames in a text file format.**

RealSenseSupport, 2020-09-17:
Thank you for highlighting an issue with the rs-convert tool with the T265 camera. We have moved our focus to our next generation of products and consequently, we will not be addressing this issue in the T265.

closed 2020-09-21 by RealSenseSupport