realsenseai/librealsense #5559: Set the resolution by advanced_mode or configure

princetonpku, 2019-12-30:
There are two ways to setting the resolution of stereo frame and color frame.
## 1、advanced_mode, the json file has the item.  Howevere, this donot work, the frame I got is the default resolution 848x480

/*  json file
...
 "stream-depth-format": "Z16",
    "stream-fps": "30",
    "stream-height": "720",
    "stream-width": "1280""
*/

	rs2::context ctx;
	std::vector<rs2::pipeline> pipes;
	rs2::device_list devices = ctx.query_devices();

	rs400::advanced_mode advanced = devices[0].as<rs400::advanced_mode>();

	std::ifstream fileJson("Default_D435_1280X720.json");
	std::string strJson((std::istreambuf_iterator<char>(fileJson)), std::istreambuf_iterator<char>());
	advanced.load_json(strJson);
        std::shared_ptr<rs2::pipeline> pipe = std::make_shared<rs2::pipeline>();

## 2 configure. This way works.
	rs2::config cfg;
	cfg.enable_stream(RS2_STREAM_COLOR, 1280, 720, RS2_FORMAT_BGR8);
	cfg.enable_stream(RS2_STREAM_DEPTH, 1280, 720, RS2_FORMAT_Z16);
	pipe->start(cfg);

I want ask the reasons and the differenc between **rs400::advanced_mode** and **rs2::config**

ev-mp, 2020-01-06:
@princetonpku hello,
The JSON preset files _control_ the `advanced_mode` attributes and do not affect the streaming configurations. The `rs2::config` object on the other hand is used to gather and accumulate the streaming environment  requests: sensors types, formats, resolutions, record settings, etc' .
Currently the functionality provided by those two constructs in the SDK do not overlap.

RealSenseCustomerSupport, 2020-01-10:
----------------------------------------------

Hi @princetonpku,

The main purpose of advanced mode presets is to provide specific configuration of advanced mode controls. As @ev-mp have noticed, for stream settings configuration one should use rs2::config object. Please let us know if you need any further clarifications regarding this topic.

Thank you!

princetonpku, 2020-01-13:
@RealSenseCustomerSupport  Thank you for your reply！I got it。

closed 2020-01-13 by princetonpku