realsenseai/librealsense #5560: T265: Feature request regarding static nodes

neilyoung, 2019-12-30:
I believe, I have raised this already somewhere, but I can't find the feature request anymore, sorry for doubling.

The current situation allows to store/retrieve a static node via `get_static_node/set_static_node`. It works fine.

It would be great to have a generic `list_static_nodes` or `enumerate_static_nodes` function, which enumerates all names of nodes stored in a given map, so that a user of that map would not be forced to know the names of the nodes beforehand. This would allow to store arbitrary and retrievable custom-information via the node's names (with respect to the limits of course), which could be used later on.

radfordi, 2020-01-08:
👍

neilyoung, 2020-01-08:
@radfordi Thanks for the thumb up

RealSenseSupport, 2020-09-23:
Hi @neilyoung ,

Thanks for the enhancement request. At this time we have moved our focus to our next generation of products and consequentially will not be addressing this feature for T265.

closed 2020-10-26 by RealSenseSupport