realsenseai/librealsense #5562: Getting / Setting ROI on MacOS 10.15.2 causing freeze

compscidr, 2019-12-31:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | { D435 and D415 } | 
| Firmware Version                   | 05.11.15.00 and 05.12.01.00 | 
| Operating System & Version |   {MacOS 10.15.2  | 
| Kernel Version (Linux Only)    |  19.2.0                                         | 
| Platform                                 |  MacOS 10.15.2  |
| SDK Version                            |  { 2.31.0 (development branch) }                          | 
| Language                            |  { Python }                          | 
| Segment			|  { Robot }                          | 

### Issue Description
Having problems setting and getting the ROI from the RGB and Depth sensors. The code is working fine when I run it on Ubuntu, however with MacOS, it freezes at the lines where I try to get or set the sensor ROI. Doesn't seem to matter if I only do one sensor, or the other, or both.

compscidr, 2019-12-31:
Was playing around with this a bit more today - if you get or set the ROI before you start the pipeline it doesn't freeze, but if you try to set the ROI after the pipeline has started, it freezes.

dorodnic, 2020-01-07:
Hi @compscidr
I'm having trouble reproducing this... Running MacOS 10.13.6 with 5.12 firmware and 2.31 SDK and the Viewer. Do you see this only in python or also when trying to use this functionality in Viewer / C++?

compscidr, 2020-01-07:
So far only on python. Will try today with C++. Viewer seems to work fine, so I assume c++ will as well.

dorodnic, 2020-01-07:
I'll try to check python as well... Perhaps something to do with python global lock... :/

compscidr, 2020-01-07:
Confirmed it is not occurring in c++

RealSenseCustomerSupport, 2020-03-04:
----------------------------------------------

Please try to upgrade to new version.
The following code works in my laptop (macOS 10.15.3, python3)

#! /usr/bin/python3
import pyrealsense2 as rs2
import time

ctx = rs2.context()
pipe = rs2.pipeline(ctx)

pp = pipe.start()

for sensor in pp.get_device().sensors:
 if sensor.is_depth_sensor():
 print("DEPTH SENSOR")
 else:
 print("RGB SENSOR")
    roi_sensor = sensor.as_roi_sensor()
 print(roi_sensor)
    time.sleep(0.1)
    roi = roi_sensor.get_region_of_interest()
    time.sleep(0.1)
 print("BEFORE SET (",roi.min_x, ", ", roi.min_y, ") - (", roi.max_x, ", ", roi.max_y, ")")
    roi.min_x = roi.min_y = 100
    roi.max_x = roi.max_y = 200
    roi_sensor.set_region_of_interest(roi)
    time.sleep(0.1)
    roi = roi_sensor.get_region_of_interest()
    time.sleep(0.1)
 print("AFTER SET (",roi.min_x, ", ", roi.min_y, ") - (", roi.max_x, ", ", roi.max_y, ")")
    time.sleep(0.1)

pipe.stop()

compscidr, 2020-03-06:
Just tried it, seems to be working - had to modify it a bit because it start the pipeline without a config - for future reference for anyone else - this is what I had to get it to work:

```
#! /usr/bin/python3
import pyrealsense2 as rs2
import time

ctx = rs2.context()
pipe = rs2.pipeline(ctx)

devices = {d.get_info(rs2.camera_info.serial_number): d for d in ctx.query_devices()}
device_id, device = devices.popitem()

rsconfig = rs2.config()
rsconfig.enable_device(str(device_id))

rsconfig.enable_stream(rs2.stream.color, 1920, 1080, rs2.format.any, 30)
rsconfig.enable_stream(rs2.stream.depth, 1280, 720, rs2.format.any, 30)
rsconfig.enable_stream(rs2.stream.infrared, 1, 1280, 720, rs2.format.any, 30)
rsconfig.enable_stream(rs2.stream.infrared, 2, 1280, 720, rs2.format.any, 30)

pp = pipe.start(rsconfig)

for sensor in pp.get_device().sensors:
    if sensor.is_depth_sensor():
        print("DEPTH SENSOR")
    else:
        print("RGB SENSOR")
    roi_sensor = sensor.as_roi_sensor()
    print(roi_sensor)

time.sleep(0.1)
roi = roi_sensor.get_region_of_interest()
time.sleep(0.1)
print("BEFORE SET (",roi.min_x, ", ", roi.min_y, ") - (", roi.max_x, ", ", roi.max_y, ")")
roi.min_x = roi.min_y = 100
roi.max_x = roi.max_y = 200
roi_sensor.set_region_of_interest(roi)
time.sleep(0.1)
roi = roi_sensor.get_region_of_interest()
time.sleep(0.1)
print("AFTER SET (",roi.min_x, ", ", roi.min_y, ") - (", roi.max_x, ", ", roi.max_y, ")")
time.sleep(0.1)

pipe.stop()
```

Thanks for the help!

closed 2020-03-06 by compscidr