realsenseai/librealsense #5563: Pose stream performance when AprilTag is used

jessearmand, 2019-12-31:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | T265 | 
| Firmware Version                   |  0.2.0.879 | 
| Operating System & Version |  Raspbian Buster  | 
| Kernel Version (Linux Only)    |  4.19.75-v7l+ | 
| Platform                                 | Raspberry Pi 4 |
| SDK Version                            | b60c383f0df4dfabd1ee56cfacf412d0c2d4ff56 | 
| Language                            | C++ | 
| Segment			| Robot | 

I have been running `rs-pose-apriltag.cpp` example on raspberry pi 4.

First of all, if I run `rs-pose.cpp` example, the frequency of pose update is very fast at 200 Hz, and this is great. But, if I run the `rs-pose-apriltag.cpp` and printing the camera pose when the april tags are not detected, the frequency significantly drops. I'm aware that the detection is done inside an `std::async` closure, and I print the camera pose outside the async closure.

Then, I tried to disable both fisheye streams after 2 april tags are detected, restart the pipeline, and the pose stream goes back to its maximum rate of 200 Hz. But, I can't do a pipeline restart in my use case, because in this case the orientation of the camera will be restarted to when it was first powered on.

Let's say just before the point of restart, the camera is oriented at 90 degrees in world coordinate, after the pipeline is restarted, the camera is oriented at 0 degrees, while it's supposed to be 90 degrees. 

Any way to mitigate this performance problem without restarting the pipeline and disabling fisheye streams? I'm already using USB 3 interface on Raspberry Pi 4. I wonder why the performance still suffers.

As a comparison, I don't have this pose stream rate slowdown while running on [Intel NUC 7i3DNKE](https://ark.intel.com/content/www/us/en/ark/products/122495/intel-nuc-kit-nuc7i3dnke.html).

jessearmand, 2020-01-03:
I figured out that it's much better to read and output the pose stream inside the frame callback with [callback API](https://github.com/IntelRealSense/librealsense/tree/master/examples/callback)

Then, the apriltag could stay in the main loop as before. Just the fisheye frames (`rs2::frameset`) inside the callback must be shared with the main loop of apriltag detection.

Let me know if there's a better way to share variables between threads. Also, I wasn't able to put the apriltag detection inside the frame callback, as that makes it more likely to have a segmentation fault because the tag manager has to be created inside the callback to calibrate the values from the fisheye frames.

RealSenseSupport, 2020-09-17:
Thank you for highlighting the issue with the pose stream performance on a Raspberry Pi . We have moved our focus to our next generation of products and consequently, we will not be addressing this in the T265.

closed 2020-09-21 by RealSenseSupport