realsenseai/librealsense #5565: How to copy rs2:frameset?

Micalson, 2019-12-31:
Hi, thank you for your nice code. And I have a question: How to copy rs2:frameset? I can not find some function to achieve it. Thank you for your reply.

Regards,

ev-mp, 2020-01-06:
@Micalson hello, the `rs::frameset` is a `shared_ptr` object with the memory resources being managed exclusively by the SDK. 
Depending on the envisaged usage you can either keep/clone* the handle, or make a deep copy of the underlying frame buffers to user-provided memory.
Note that the internal memory pools are limited by default to less than 20 frames, so in case you want to keep more than 5-10 handles you must use the `frame.keep()` API to prevent starvation and `no frames arrived` (see [example](https://github.com/IntelRealSense/librealsense/blob/master/tools/benchmark/rs-benchmark.cpp#L267))

Also check #2272

Micalson, 2020-01-07:
@ev-mp I get it. Thank you for your reply.

closed 2020-01-10 by RealSenseCustomerSupport