realsenseai/librealsense #5566: segmentation fault: pyrealsense2.so

Scientifeet, 2019-12-31:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model | SR300 | 
| Operating System & Version | Linux raspberrypi 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux  | 
| Platform | Raspberry Pi |
| Language | python | 

## Error segmentation fault
Hello, i have an problem on my script  
**file: scan.py**

> Structure of my folder  
> **/home/pi/scan**
> * scan.py
> * pyrealsense2.so
> * librealsense2.so

``` python
## License: Apache 2.0. See LICENSE file in root directory.
## Copyright(c) 2015-2017 Intel Corporation. All Rights Reserved.

#####################################################
## librealsense tutorial #1 - Accessing depth data ##
#####################################################

# First import the library
import pyrealsense2 as rs

ctx = rs.context()
if len(ctx.devices) > 0:

	for d in ctx.devices:

	    print ('Found device: ', \

	            d.get_info(rs.camera_info.name), ' ', \

	            d.get_info(rs.camera_info.serial_number))

else:

    print("No Intel Device connected")
```

When i apply this script in my termina with that command:
``` bash
python3 scan.py
```

I have that error **<Segment fault (core dumped)>**
You are an solution ?
Regards

RealSenseCustomerSupport, 2020-02-04:
----------------------------------------------

@Scientifeet Can you run rs-enumerate or rs-depth successfully on your setup? And did you get chance to try python examples such as https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/python-tutorial-1-depth.py to see if any issue?

RealSenseCustomerSupport, 2020-02-13:
----------------------------------------------

@Scientifeet Any update from your side? Looking forward to your reply. Thanks!

RealSenseCustomerSupport, 2020-02-25:
----------------------------------------------

@Scientifeet Could you please update?

RealSenseCustomerSupport, 2020-03-04:
----------------------------------------------

@Scientifeet  Sorry we didn't get response from you for weeks, will close the issue at this point. Please feel free to create new one if you still have issues or questions.Thanks!

closed 2020-03-04 by RealSenseCustomerSupport

chapa17, 2021-08-24:
Hello,

I am also getting a similar error when using python3.7 on the code line **pipeline.start(config)**. Can you please help me with this?