realsenseai/librealsense #5567: Cannot build on old macOS: use of undeclared identifier 'EVFILT_EXCEPT'

Gnimuc, 2019-12-31:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | N/A | 
| Firmware Version                   |  N/A | 
| Operating System & Version |  MacOS  | 
| Kernel Version (Linux Only)    |  10.10                                        | 
| Platform                                 | PC  |
| SDK Version                            | 2.31.0                         | 
| Language                            |  C                          | 
| Segment			|   N/A                       | 

### Issue Description
I'm trying to build librealsense in a cross-compile environment [where the default macOS version is 10.10](https://github.com/JuliaPackaging/Yggdrasil/pull/352#issuecomment-569944098). However, libtm uses `EVFILT_EXCEPT` at this line:

https://github.com/IntelRealSense/librealsense/blob/bdce5a4ebbd716e9d7372ad8831a929bccccd0b2/third-party/libtm/libtm/src/infra/Poller_bsd.cpp#L35

which is only available in macOS 10.13+.

Gnimuc, 2019-12-31:
The following filter types are defined in my system(10.15)'s  `<sys/event.h>`:
```
/*
 * Filter types
 */
#define EVFILT_READ             (-1)
#define EVFILT_WRITE            (-2)
#define EVFILT_AIO              (-3)    /* attached to aio requests */
#define EVFILT_VNODE            (-4)    /* attached to vnodes */
#define EVFILT_PROC             (-5)    /* attached to struct proc */
#define EVFILT_SIGNAL           (-6)    /* attached to struct proc */
#define EVFILT_TIMER            (-7)    /* timers */
#define EVFILT_MACHPORT         (-8)    /* Mach portsets */
#define EVFILT_FS               (-9)    /* Filesystem events */
#define EVFILT_USER             (-10)   /* User events */
#define EVFILT_VM               (-12)   /* Virtual memory events */
#define EVFILT_EXCEPT           (-15)   /* Exception events */

#define EVFILT_SYSCOUNT         17
#define EVFILT_THREADMARKER     EVFILT_SYSCOUNT /* Internal use only */
```
I'm not sure why only `EVFILT_EXCEPT` is missing in macOS 10.10's  `<sys/event.h>`. 

I copy-pasted `#define EVFILT_EXCEPT  (-15)` to `Poller_bsd.cpp` and now it can compile successfully. But is this a decent fix?

dorodnic, 2020-01-01:
Hi @Gnimuc 
If you download the latest development branch, you should no longer see the problem. 
We are actually in process of removing "libtm" all-together, and I don't think the new streamlined implementation should suffer from this problem. Let us know

Gnimuc, 2020-01-02:
Thanks for the info. I can successfully build the latest development branch.

closed 2020-01-02 by Gnimuc