realsenseai/librealsense #5568: D435i depth is not accurate

diyuhuodong, 2019-12-31:
librealsense | 2.30.0 RELEASE
-- | --
OS | Mac
Name | Intel RealSense D435i
Firmware Version | 05.11.11.100

I checked the accuracy of D435I with the tool rs-depth-quality and found that there is always an error when compared with the actual distance. For example, the actual distance is 1171 cm, and the measurement result of D435i is 1130 cm. The result is 41 cm away, which is a bit large.
[image]

I used the D415 to do the same test and found that the distance measured was accurate.

[image]

I don't know if this is a hardware issue or a setup issue.

dorodnic, 2020-01-01:
Hi @diyuhuodong 
Z-accuracy metric, like other depth metrics, are affected by individual camera calibration. While every camera is calibrated during manufacturing, physical disturbances during shipping can cause degradation. I'd speculate this can be the case here. 
We provide two methods of addressing this issue:
a. Under camera menu (the "hamburger icon" under camera name in the Viewer / DQT) there is On-Chip and Tare Calibration options. In your case you are interested in Tare Calibration, as it is designed to improve z-accuracy. You will be prompted to provide ground-truth (1171) and the algorithm will run for couple of seconds. You will be able to compare before / after calibration and make a decision if you are satisfied with algorithms output. 
b. You can follow the documentation on our site to download and install Dynamic Calibration tool. This method would require more time and effort, but offers a reliable plan B in case SDK algorithms do not work as expected

diyuhuodong, 2020-01-03:
Thank you for replying, I will calibrate the camera separately in two ways today, and I will return the test results

diyuhuodong, 2020-01-05:
My problem is solved, thank you very much.@dorodnic

closed 2020-01-05 by diyuhuodong