realsenseai/librealsense #5569: More detail into how Corrupted Depth Image Firmware Issue (DSO-11040 from the latest spec sheet) was fixed

fhenrywells, 2019-12-31:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D400  | 
| Firmware Version                   | 05.11.01.00 | 
| Operating System & Version |  Linux (Ubuntu 16) | 
| Kernel Version (Linux Only)    |  4.15.0-72-lowlatency      | 
| Platform                                 | PC |
| SDK Version                            |  Not sure                       | 
| Language                            |  Python                           | 
| Segment			|  Robot                     | 

### Issue Description
 Issue DSO-11040 from the latest RealSense D-435 spec sheet (https://www.intel.com/content/dam/support/us/en/documents/emerging-technologies/intel-realsense-technology/RealSense-D400-Series-Spec-Update.pdf) 

I am curious as to how this problem was resolved as I suspect the bug occurred in one of our deployments running old firmware. On a related note, how can the autoexposure algorithm reduced the exposure levels to below -6?

fhenrywells, 2020-01-06:
It turns out that this issue was not resolved, since it emerged on a D415 camera, not D435. Here is a sample depth map that is corrupted, curious to know if this is something that has been seen before:
[image]

RealSenseCustomerSupport, 2020-01-13:
----------------------------------------------

Hi @ fhenrywells
(1) as to the question about -6 exposure level,
you can refer to
 https://github.com/IntelRealSense/librealsense/issues/5493
the -6 refer to the MS logarithmic scale :
Just to explain with given example - the value that you read is 625. If you inspect the Librealsense's exposure control description then you'd find that it reports 0.1msec units, i.e 625=>>62.5 msec, and 62.5 msec corresponds to 1/16 of a seconds, or (-4) using MS logarithmic scale.
-6 =1/64 =0.015625 which corresponding to  1-2 of RGB exposure time
means the exposure time could be  to set to very close to 0
this is helpful under  the environment with very  strong light

(2) the issue DSO-11040 likely not the same issue you have now (D415 bad depth map you post)

- Please see if use the latest FW version with latest Librealsense API  can help
the easier way is to download the latest viewer (v2.3x) that use the latest API and also can use it to update the FW
you can run the latest version of viewer to see if the issue still there first.

-how many camera streaming at the same time on the same host?
it looks the resolution you set is the highest  1289x720..

there some limitation on the fps/resolution if you want to stream multi-camera at the same time due to the USB3 bandwidth limitation  that host able to deal with

to check if this matter, you can lower the resolution ex :640x480
as to the limitation, you can found from the white paper, link provided below

https://simplecore.intel.com/realsensehub/wp-content/uploads/sites/63/Multiple_Camera_WhitePaper04.pdf

closed 2020-02-24 by RealSenseCustomerSupport