realsenseai/librealsense #5575: Depth Sensor ROI

Jiloc, 2020-01-03:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  D400 | 
| Language                            |  C++                   | 

Hello,

Is there something available to set a ROI for the depth stream? I know there is a threshold filter, but it can set only the min and max Z, not X and Y. It's not a big deal to do it manually but I just wanted to know if there is some standard way in the SDK.

dorodnic, 2020-01-03:
Hi @Jiloc 
Not something built-in. I think newer firmware versions have 848x100 and another small resolution that are essentially cropped from the native resolution in order to allow higher framerates.

Jiloc, 2020-01-03:
as always, thank you

closed 2020-01-03 by Jiloc