realsenseai/librealsense #5578: About  spatial position？？？

CharlesPikachu, 2020-01-03:
Is there an api that maps pixels to spatial locations？
Could anyone give me an example？
Thanks a lot！！！

MartyG-RealSense, 2020-01-03:
It sounds as though you are asking about mapping 2D pixel coordinates to 3D world coordinates?  If so, this is a process called Deprojection, described by the Librealsense instruction **RS2_DEPROJECT_PIXEL_TO_POINT**

I hope that the long discussion in the link below, which has Python scripts, will be useful to you.

https://github.com/IntelRealSense/librealsense/issues/1904

dorodnic, 2020-01-07:
I'm sorry, the question is not clear. Hopefully Marty's suggestion is relevant, if not please open a new issue with detailed description

closed 2020-01-07 by dorodnic

CharlesPikachu, 2020-01-08:
thank you for the reply，I have found the solution according to your guide. thanks again for your helps.