realsenseai/librealsense #5581: How to set the camera's auto exposure parameters?

Dontla, 2020-01-04:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435| 
| Firmware Version                   | 05.11.06.250| 
| Operating System & Version |  Win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.25.0                     | 
| Language                            |  python3.6                           | 
| Segment			|  Robot                       | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I am currently using the camera's auto exposure mode, but when encountering abnormal light, auto exposure can not achieve good results, so I want to reset the camera's auto exposure mode. How can I achieve this in code?
Best regard

MartyG-RealSense, 2020-01-04:
The common approach when a problem with the auto-exposure occurs due to a lighting event is to reset the entire camera.  This can be done in code with a **hardware_reset()** instruction.

Some users program their scripts to only initiate the reset when certain conditions are met.  For example, @SirDifferential based a reset condition on the number of zero pixels currently in the image.

https://github.com/IntelRealSense/librealsense/issues/5209#issuecomment-563234576

If you prefer to turn the auto-exposure off and then turn it on again, the link below has a code sample for doing so with the **RS2_OPTION_ENABLE_AUTO_EXPOSURE** instruction.  

https://github.com/IntelRealSense/librealsense/issues/3141#issuecomment-457361895

Setting it to '0' disables auto-exposure and setting it to '1' enables it.

Dontla, 2020-01-07:
How can I set the parameters of auto exposure?

For example, I think AE is not bright enough, every time I want it brighter, how to do it?

Best regard!

Dontla, 2020-01-07:
Then, I thought, is there any python code to set the exposure value?

Converting c code to python code is too difficult for me....

MartyG-RealSense, 2020-01-07:
To set manual exposure values, you have to first turn auto-exposure off.

I hope the link below will be helpful to you in setting the exposure value with Python.

https://github.com/IntelRealSense/librealsense/issues/4449

RealSenseCustomerSupport, 2020-01-09:
----------------------------------------------

there are two way for exposure
-disable the auto exposure then setup the exposure time
-enable auto exposure then adjust the mean intensity set point
which is under Viewer->advance controls->AE control -> mean intensity set point
the larger value the brighter image.
would suggest you to use viewer to do some test under the lighting condition you are interested to find the proper value.

closed 2020-02-24 by RealSenseCustomerSupport