realsenseai/librealsense #5582: C# Align to Color Frame and Finding Depth

AikakiAdmin, 2020-01-04:
I am using the following code, I have seen python code and C++ and in all the example code. align() returns FrameSet i.e people are able to get depth frame and color frame. In C# it is returning a processed frame of type Fame. I want to call getdistance(x,y) function but I can't use it. How I calculate distance (depth) using color frame coordinate?

using(var ctx = new Context())
                {
                        var devices = ctx.QueryDevices();

                        Console.WriteLine("There are {0} connected RealSense devices.", devices.Count);
                        if (devices.Count == 0) return 1;
                        var dev = devices[0];

                        Console.WriteLine("\nUsing device 0, an {0}", dev.Info[CameraInfo.Name]);
                        Console.WriteLine("    Serial number: {0}", dev.Info[CameraInfo.SerialNumber]);
                        Console.WriteLine("    Firmware version: {0}", dev.Info[CameraInfo.FirmwareVersion]);
                 }

              var cfg = new Config();
                cfg.EnableStream(Intel.RealSense.Stream.Depth, desiredFrameWidht, desiredFrameHeight);
                cfg.EnableStream(Intel.RealSense.Stream.Color, desiredFrameWidht, desiredFrameHeight, Format.Bgr8);
                

                Console.WriteLine("Starting Realsense pipeline");
                var pipe = new Pipeline();
                var pp = pipe.Start(cfg);
                
                Console.WriteLine("After pipeline start");
                Colorizer colorizer = new Colorizer();
                Align align = new Align(Intel.RealSense.Stream.Color)
for (;;)
				{
					

                    using (var frames = pipe.WaitForFrames())
                    using (var colorframe = frames.ColorFrame)
                    using (var depthframe=frames.DepthFrame)
                    {
                                      
                        var aligned_frame = align.Process(frames);
                        var colorized_aligned_frame = colorizer.Process(aligned_frame);

                           OpenCvSharp.Mat frame=new OpenCvSharp.Mat(desiredFrameHeight,desiredFrameWidht,OpenCvSharp.MatType.CV_8UC3,colorframe.Data,0);
                        OpenCvSharp.Mat dframe = new OpenCvSharp.Mat(desiredFrameHeight, desiredFrameWidht, OpenCvSharp.MatType.CV_8UC3, colorized_aligned_frame.Data, 0);
        
                        // give a frame to the worker
                        worker.work(frame,dframe);

MartyG-RealSense, 2020-01-04:
The C# documentation has sample code for getting distance from the depth frame with GetDistance.  I wonder if you would get the result that you needed if you placed this distance measuring code after the alignment of depth to color.

https://github.com/IntelRealSense/librealsense/tree/master/wrappers/csharp#hello-world

AikakiAdmin, 2020-01-04:
Issue is after we do alignment
 var aligned_frame = align.Process(frames);

returned object is of type Frame, to call GetDistance() frame has to be DepthFrame. It gives error at compile time. No way I get DepthFrame type after align.Process in C#  and I have searched all C++ and Python example and it appears interface is different. I have a working code in Python but in C# I am can't move forward. following code worked well in python

  # Get frameset of color and depth
        frames = pipeline.wait_for_frames()
        # frames.get_depth_frame() is a 640x360 depth image
        
       
        # Align the depth frame to color frame
        aligned_frames = align.process(frames)

        # Get aligned frames
        aligned_depth_frame = aligned_frames.get_depth_frame() # aligned_depth_frame is a 640x480 depth image
        color_frame = aligned_frames.get_color_frame()

        # Validate that both frames are valid
        if not aligned_depth_frame or not color_frame:
            continue
        
        depth_image = np.asanyarray(aligned_depth_frame.get_data())
        color_image = np.asanyarray(color_frame.get_data())

MartyG-RealSense, 2020-01-04:
I have seen a few reports in the past 18 months about users experiencing problems with aligning depth to color in C#.  Here's an example case that illustrates this.

https://github.com/IntelRealSense/librealsense/issues/4719

AikakiAdmin, 2020-01-04:
> I have seen a few reports in the past 18 months about users experiencing problems with aligning depth to color in C#. Here's an example case that illustrates this.
> 
> #4719

I have reference the code and I have a simple code not using call backs, I have used the code mentioned in #4719. it gives me both aligned blocks but they are still not proper ColorFrame and DepthFrame (They are frames).   I need to call this AllocatedComposite() as mentioned in the code but I am not sure where to get the FrameSource object.  I just created one to try crashes. Any idea? I am just wonder where this FrameSource is coming from.

                       var aligned_frames=frames.ApplyFilter(align);
                        aligned_frames = aligned_frames.ApplyFilter(colorizer);
                        var aligned_frameset = aligned_frames.As<FrameSet>();
                        
                        
                        var colorFrame = aligned_frameset[Intel.RealSense.Stream.Color, Format.Bgr8];
                        var AligneddDepth = aligned_frameset[Intel.RealSense.Stream.Depth, Format.Rgb8];

AikakiAdmin, 2020-01-06:
This is still open, see my last reply

MartyG-RealSense, 2020-01-06:
I considered this question further but it is outside of my programming knowledge, so one of the Intel RealSense team members on this forum will be better equipped to answer it.  I'll add a couple of tags to bring it to their attention.

@ev-mp  @dorodnic

mjyn, 2020-03-07:
Same problem here.
I found these lines in Align.cs
```c#
[Obsolete("This method is obsolete. Use Process with DisposeWith method instead")]
public FrameSet Process(FrameSet original, FramesReleaser releaser)
{
    return Process(original).As<FrameSet>().DisposeWith(releaser);
}

so i tried 
```c#
Align align = new Align(Stream.Color);
FrameSet alignedframeset= align.Process(frameset).As<FrameSet>().DisposeWith(frameset);

It works, but after several frames. I got an exception: “Frame didn't arrived within 5000”

MartyG-RealSense, 2020-03-07:
@mjyn   I hope the link below will be of help to you in finding working C# align code.

https://github.com/IntelRealSense/librealsense/issues/4719#issuecomment-526046153

MartyG-RealSense, 2020-03-07:
@mjyn  To help avoid the frames din't arrive within 5000 error, C# users should ensure that their code is wrapped in a Using block as well as using dispose (as you have done).

https://github.com/IntelRealSense/librealsense/issues/2845#issuecomment-444448923

mjyn, 2020-03-07:
> @mjyn To help avoid the "frames didn't arrive within 5000" error, C# users should ensure that their code is wrapped in a Using block as well as using dispose (as you have done).
> 
> [#2845 (comment)](https://github.com/IntelRealSense/librealsense/issues/2845#issuecomment-444448923)

Seems like the Frame object before calling As<T> should be disposed to avoid the exception.

```c#
using (var frames = pipeline.WaitForFrames())
{
    Align align = new Align(Stream.Color).DisposeWith(frames);
    Frame aligned = align.Process(frames).DisposeWith(frames);
    FrameSet alignedframeset = aligned.As<FrameSet>().DisposeWith(frames);
    var colorFrame = alignedframeset.ColorFrame.DisposeWith(alignedframeset);
    var depthFrame = alignedframeset.DepthFrame.DisposeWith(alignedframeset);
}
```
My code works now, thank you.

MartyG-RealSense, 2020-03-07:
@mjyn   Great news, thanks for the update and for sharing your C# code with the community.   :)

closed 2020-12-14 by RealSenseSupport