realsenseai/librealsense #5583: rs2_transform_point_to_point

scottrpaterson, 2020-01-04:
Camera Model  - D435i
Operating System & Version - Windows 10
Language - Python

### Issue Description

In the documentation (https://intelrealsense.github.io/librealsense/python_docs/_generated/pyrealsense2.html) I'm not seeing what parameters the function rs2_transform_point_to_point accepts. Is there better documentation somewhere?

How do I use this function? I'm trying to move my viewpoint to the end of my robot arm end effector.

MartyG-RealSense, 2020-01-04:
It is not an instruction I have used myself, but I found a Python script that makes use of it to act as a reference for you.

https://github.com/IntelRealSense/librealsense/issues/1231#issuecomment-368421888

[image]

The 'Extrinsic Camera Parameters' of the Librealsense SDK's projection documentation also provides some explanation about use of rs2_transform_point_to_point 

https://dev.intelrealsense.com/docs/projection-in-intel-realsense-sdk-20#section-extrinsic-camera-parameters

This documentation section in turn links to another code example for use of the instruction.

https://github.com/IntelRealSense/librealsense/blob/5e73f7bb906a3cbec8ae43e888f182cc56c18692/include/librealsense2/rsutil.h#L68

RealSenseCustomerSupport, 2020-01-08:
----------------------------------------------

Hi @scottrpaterson,

rs2_transform_point_to_point function transforms points from one coordinate space to another.
The function accepts following parameters :
   - "rs2_extrinsics" struct that contains parameters describing relationship between the separate 3D coordinate systems.
   - 3D point coordinates.

Please let us know if you need any further assistance with this matter.
Thank you!

closed 2020-01-15 by RealSenseCustomerSupport