realsenseai/librealsense #5587: hardware accelerated depth encoding and depth units

bmegli, 2020-01-05:
I have recently shared example of hardware accelerated Realsense (D400) depth encoding to HEVC Main10 profile:

[realsense-depth-to-vaapi-hevc10](https://github.com/bmegli/realsense-depth-to-vaapi-hevc10)

The example encodes 16 bit depth Realsense data in 10 bits of HEVC Main 10 with VAAPI [P010LE](https://github.com/bmegli/hardware-video-encoder/issues/18#issuecomment-569501602) pixel format. 

The aim is to have a purely hardware based pipeline (no depth data processing on the host). This is achieved by using Realsense `Depth Units` in such way that Realsense outputs P010LE compatible data with controlled precision/range trade-off. 

The current limits of Realsense `Depth Units` are 0.0001 - 0.01, as is reported by realsense-viewer or `depth_sensor.get_option_range(RS2_OPTION_DEPTH_UNITS);`

Those limits, when translated to P010LE mean:
- the best precision/worst range is 6.4 mm/6.5535 m (for depth units 0.0001)
- the worst precission/best range is 64 cm/655.35 m (for depth units 0.01)
- all trade-offs in between are possible with purely hardware based encoding pipeline

My question here is: 

Are the limits of Realsense `Depth Units` (0.0001, 0.01) based on some hardware constraints? Would it be possible to relax them? Extending lower bound to 0.00001 would allow control of precision/range trade-off starting from 0.64 mm/0.65 m (precision/range).

--------------------------------------------------------------------

Here I include explanation of encoding on bit-level. This is not needed to answer the question, just for the "interested reader".

 [P010LE](https://github.com/bmegli/hardware-video-encoder/issues/18#issuecomment-569501602)  pixel format is like NV12, this means Y luminance plane followed by interleaved U/V plane half the size of Y. The color data is encoded in 16 bit words of which only 10 MSB is used. For depth encoding we are only interested in Y luminance plane, the U/V plane will be filled with constant values.

The P010LE pixel format was designed to be binary compatible with P016LE, which encodes Y data with 16 bits. P010LE may be passed in place of P016LE (6 LSB as zeroes). P016LE may be passed as P010LE (6 LSB dropped, this means dropping precision of data).

We may consider Realsense Z16 depth data as P016LE Y data. When passing it to VAAPI hardware encoder as P010LE we are dropping the 6 LSB bits of precision. This is why we get 6.4 mm/6.5535 m precision/range for depth units 0.0001.

-------------------------------------------------------------------------

Related to #4819, #3877, #3665, #3259

For anybody planning to use similar approach it is important that he/she understands implications of using video codec for depth encoding.

If you are not constrained by CPU and real-time requirements consider using [HEVC 3D extension](https://hevc.hhi.fraunhofer.de/3dhevc) software encoder designed for compressing depth data or other approach.

agrunnet, 2020-01-06:
There is no Hardware restriction.  We can extend it to less than 0.1. Will do so shortly in one of next FW updates.

bmegli, 2020-01-06:
Hi @agrunnet,

Just to make it clear - what is needed is extension on the precision side (0.0001 to 0.00001)

When the data is passed to VAAPI for hardware encoding 6 LSB bits of precision are lost

| Side |       |       |     |     |   |    |     |    |     |   |    | Side |
|-------- |-----|-----|----|----|---|---|----|---|----|---|---|--------|
| MSB | 15 | 14 | ... | 7 | 6 | 5 | 4 | 3 | 2  |1 | 0 | LSB |

So the bits 5-0 are completely ignored by the VAAPI hardware. This means that we have to configure Realsense to output:
- 6 bits more precise data than we need in the application
- in other words 2^6=64 times more precise

This is also why we are loosing range in the process.

0.00001 depth unit after encoding would translate to:
- 0.00001 * 64 = 0,00064 m = 0.64 mm precision
- 0.00001 * 65472 = 0.65472 m range (65472 in binary is 10 ones followed by 6 zeroes)

Regards

agrunnet, 2020-01-06:
As you say, if you set depth units to 1000 then each step is 1m and max range is 65m. If you set it to 10 then the steps are 10um and the max range is 0.65m. 
So if you operate the D435 at 848x480 where the minimum distance is 17cm, then depth would only appear in range of 17-65cm. Below 17cm things would be black (zero) and above it would be clamped at 65cm. 
Is this what you want?
In any event we are changing the allowable range for the depth units so you can go down to 1um steps in the future. I believe from FW 5.12.1.0

bmegli, 2020-01-06:
> if you set depth units to 1000 then each step is 1m and max range is 65m.

... is 1 mm and max range is 65 m (typo m -> mm)

> If you set it to 10 then the steps are 10um and the max range is 0.65m.

Ok, I understand it now. You are speaking in different units (number of um, internal device units?), I in user interface units (fraction of m)

|                 | realsense viewer controls->depth units  | realsense viewer advanced controls->depth table->depth units| Realsense API  | json with settings | internal device side|
|------------- |-------------------------|--------------------|------------------------|-------------------------|-----------------------------|
| example |       0.001              |     1000           |  0.001              |          1000            |   something                |

> [...] Is this what you want?

Generally yes. I know it could be made better with offset for minimal range.

> In any event we are changing the allowable range for the depth units so you can go down to 1um steps in the future. I believe from FW 5.12.1.0

Awesome! This is even more than is needed.

If it is not a problem I will keep the issue open until the firmware is released.

Kind regards

RealSenseCustomerSupport, 2020-01-13:
----------------------------------------------

Hi @bmegli FW 5.12.1.0 is released. Do you have a chance to give it a shot?
Thanks,

bmegli, 2020-01-13:
Hi,

With Development Firmware FW 5.12.1.0 I confirm that it is possible to set `realsense viewer controls->depth units` in range 0.000001-0.01

Thanks!

closed 2020-01-13 by bmegli