realsenseai/librealsense #5588: Announcement:  Intel reveal powerful Ghost Canyon NUC at CES 2020

MartyG-RealSense, 2020-01-06:
Hi everyone, 

Intel have unveiled the new Ghost Canyon NUC at CES 2020.  It is highly upgradable, and offers up to 5Ghz on a single core and room for a desktop grade GPU.

https://www.engadget.com/2020/01/06/intel-ghost-canyon-nuc-desktop-gpu/

dorodnic, 2020-01-07:
Nice, but also not strictly relevant to RealSense :)

closed 2020-01-07 by dorodnic

MartyG-RealSense, 2020-01-07:
With the greatest of respect, I carefully consider announcements for relevance to RealSense.  NUC has a  user-base in robotics (very relevant to RealSense ROS users).  So the story was relevant in that regard.

dorodnic, 2020-01-07:
I apologise, you are 100% welcomed to share and contribute. Just trying to get issues under control and make sure customer reports don't get pushed to far. 
With CES on-going, there are many noteworthy news. Perhaps we could group all relevant technology news into a single page of CES recap? What do you think?
Thanks @MartyG-RealSense

MartyG-RealSense, 2020-01-07:
I'd already adopted a policy of no more than two Announcements on the front page a while ago, and close one of them early to make way for a new one if a new story emerges whilst there are two on the front page.  I also periodically do batch closures of older Announcements to reduce the issue count if a RealSense team member has not closed them already.

The non-Intel CES news relatable to RealSense is probably mostly over now, as companies put out their best news on the day before CES or opening day in a rush for the headlines.

Thanks @dorodnic