realsenseai/librealsense #5591: Anyone going to be at CES showing off new camera

chicagoBob123, 2020-01-06:
Would like to see where the new L515 is at but won't get in until Wednesday. So I was wondering if it would be displayed in any of the many Intel booths or if Realsense engineers would be there. 
Thanks,
Bob

MartyG-RealSense, 2020-01-08:
I am sorry that you did not receive an official response in time for your visit.  At this point, the best advice may be to not make a special trip to the Intel booth in search of RealSense information if you were not planning on otherwise visiting the stand.  There is no report about what was revealed at the Tuesday evening RealSense reception event at the time of writing this.

RealSenseCustomerSupport, 2020-01-09:
----------------------------------------------

Yes. RealSense will demo L515 on CES.

closed 2020-01-14 by RealSenseCustomerSupport

chicagoBob123, 2020-01-14:
Thanks but its sad you can search for CES and Intel and NOT find something this.

MartyG-RealSense, 2020-01-14:
Some additional L515 info was distributed during CES via the RealSense Twitter and Facebook accounts after the Cubemos item listed above.  I can appreciate your disappointment though regarding not being able to network in person with a RealSense representative at CES though as you didn't arrive til the day after the reception evening.

**Full-body 3D person capture with multiple L515 being used simultaneously.**

https://twitter.com/IntelRealSense/status/1215110449169076224

chicagoBob123, 2020-01-14:
That's pretty interesting.