realsenseai/librealsense #5592: using multiprocessing with realsense D435 in python

RZz96, 2020-01-06:
Hi 
I'm using the multiprocessing in python. 

process1 = multiprocessing.Process(target=Func1,arg=depth_frame,))
process1.start()
process1.join()

at process1.start(), i get the exception error : 
can't pickle pyrealsense2.pyrealsense2.depth_frame objects

thanks in advance for the help

RealSenseCustomerSupport, 2020-02-03:
----------------------------------------------

Please have a try this sample.
https://github.com/PINTO0309/MobileNet-SSD-RealSense/blob/master/MultiStickSSDwithRealSense.py

RealSenseCustomerSupport, 2020-02-26:
----------------------------------------------

@RZz96 Did you get chance to try? Any other questions about this? Looking forward to your update. Thanks!

RZz96, 2020-02-26:
> 
> 
> @RZz96 Did you get chance to try? Any other questions about this? Looking forward to your update. Thanks!

No i couldnt use it in windows

RealSenseCustomerSupport, 2020-03-04:
----------------------------------------------

I can get the correct frame numbers with multiprocessing library. (Win10+Python 3.7)
https://gist.github.com/mengyui/8d00dbbefff65e70db9691965b99efb8
You can have a try in command line.

Please double check you have a clean environment.

RealSenseCustomerSupport, 2020-04-08:
----------------------------------------------

@RZz96 Sorry that we didn't get response from you for weeks. Will close it at this point. Please feel free to create new one if you still have questions or issues. Thanks!

closed 2020-04-15 by RealSenseCustomerSupport

eweimholt, 2020-11-25:
@RealSenseCustomerSupport Do you have any examples of multiprocessing in python for D415/435 without using the Intel Movidius Neural Compute Stick?