realsenseai/librealsense #5593: 'abs': ambiguous call to overloaded function

gpokhark, 2020-01-06:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {D345} | 
| Firmware Version                   | (05.11.15.00) | 
| Operating System & Version |   {Win (10)   | 
| Kernel Version (Linux Only)    |                                          | 
| Platform                                 | PC  |
| SDK Version                            |  { legacy / 2.30 }                          | 
| Language                            |  {C++}                          | 
| Segment			|  {others }                          | 

### Issue Description
<Describe your issue / question / feature request / etc..>

I am trying to modify the rs-measure.cpp example. When I add following line to the code it starts giving me errors. 
`#include "cv-helpers.hpp"`
It gives me following error-
|Severity|	Code     |	Description |	Project |	File |	Line |	Suppression State|
|------------|------------|------------|------------|------------|------------|------------|
|Error|	C2668|	'abs': ambiguous call to overloaded function|	rs-measure|	C:\xx\xx\xx\Intel_Realsense\librealsense-2.30.0\build\include\librealsense2\rsutil.h|	96|	 |

I want to modify the measure code to include opencv and the helper function so that I can directly use `DEPTH_UNITS` and `depth_frame_to_meters` in order to remove the bottleneck of `get_distance` function.

Can you please explain how do I fix this?

dorodnic, 2020-01-07:
We want most examples to run without dependency on opencv. 
I suspect the error you are getting is not strictly related to realsense includes, and it only occurs because of what you are including **before** including `rsutil.h`. Please try putting the realsense-related stuff above other includes, and feel free to share the modified code. I don't think any of our code overloads [`abs`](http://www.cplusplus.com/reference/cstdlib/abs/)

gpokhark, 2020-01-08:
Thank you Sergey for the quick reply, I appreciate it. I figured out that when the order of the `#include <librealsense2/rs.hpp>` and `#include <librealsense2/rsutil.h>` is changed I get the aforementioned error. It happened because I had .clang-format which automatically formatted the cpp file and then caused this issue. If you would like to have a look at my code please have a look at this [repo](https://github.com/gpokhark/Librealsense_Personal/blob/master/07_measure_depth_units/rs-measure.cpp). I have modified the existing `measure` example to calculate the distance using the `DEPTH_UNITS` and `depth_frame_to_meters` in order to remove the bottleneck of `get_distance` function.

I would appreciate if you could let me know if I have implemented it correctly or not? Because I am not getting the same measurement compared to the original example. Thank you sir for your time and effort.

RealSenseCustomerSupport, 2020-01-14:
----------------------------------------------

Hi @gpokhark,

It would be very helpful if you could point on specific problem in librealsense code that causes, in your opinion, to your code not to get "the same measurement compared to the original example".

Thank you!

closed 2020-01-28 by RealSenseCustomerSupport