realsenseai/librealsense #5595: RGB+D streams reading is around 6 FPS, could this be faster?

YuzheHao, 2020-01-07:
[device]: D435i

I am using the example program "save-to-disk" to read the RGB frame and DEPTH frame at the same time, but the reading speed is too slow.

Is there anyway I could read data stream faster?

dorodnic, 2020-01-07:
Hi @YuzheHao 
You can read the data of the sensor at 30 and 60 frames per second (for depth exclusively it goes even higher). What you do with the data after receiving it is up to you and your application. Save to disk can be bottle-necked by the PNG writer or disk write speed (I would still check that you are building the example and the SDK with optimizations enabled)
One faster way to record data to disk is using the recorder (please check the rs-playback-record example). Even then, in order to handle the full bandwidth of the camera you need an SSD drive, on a regular hard-drive you will loose some frames (just based on the disk write speeds vs frame sizes)

yanxh97, 2020-01-07:
Creating new threads to write png may help because IO operations are really time consuming. Something like:
```
std::vector<std::thread> threads;
while(1)
{
   //get frame
  threads.emplace_back(...);    //... refers to the write function & data
}
for (auto&& t : threads) t.join();
```

YuzheHao, 2020-01-08:
### 1. Saving method in `rs-save-to-disk`
The saving method in my example program is this:
```
stbi_write_png(png_file.str().c_str(), vf.get_width(), vf.get_height(),
               vf.get_bytes_per_pixel(), vf.get_data(), vf.get_stride_in_bytes());
```
Is this taking too much time that makes the later frames could not be reviced on time?

### 2. Try for `rs-record-playback`

And I try to run "rs-record-playback.cpp" as @dorodnic said, but it failed and the debug report is like below. some reference like `gluPerspective`,`gluLookAt`,`gluPerspective` could not be found. Whether this means I still have some package that forget to intall?
```
====================[ Build | rs-record-playback | Debug ]======================
/work/yuzhe/apps/clion-2019.3.1/bin/cmake/linux/bin/cmake --build /work/yuzhe/librealsense/cmake-build-debug --target rs-record-playback -- -j 4
[  1%] Built target fw
[ 12%] Built target glfw
[ 22%] Built target tm
[ 35%] Built target realsense-file
[ 96%] Built target realsense2
[ 98%] Linking CXX executable rs-record-playback
CMakeFiles/rs-record-playback.dir/rs-record-playback.cpp.o: In function `draw_pointcloud(float, float, glfw_state&, rs2::points&)':
/work/yuzhe/librealsense/examples/record-playback/../example.hpp:773: undefined reference to `gluPerspective'
/work/yuzhe/librealsense/examples/record-playback/../example.hpp:777: undefined reference to `gluLookAt'
CMakeFiles/rs-record-playback.dir/rs-record-playback.cpp.o: In function `draw_pointcloud_wrt_world(float, float, glfw_state&, rs2::points&, rs2_pose&, float*, std::vector<rs2_vector, std::allocator<rs2_vector> >&)':
/work/yuzhe/librealsense/examples/record-playback/../example.hpp:839: undefined reference to `gluPerspective'
collect2: error: ld returned 1 exit status
examples/record-playback/CMakeFiles/rs-record-playback.dir/build.make:134: recipe for target 'examples/record-playback/rs-record-playback' failed
make[3]: *** [examples/record-playback/rs-record-playback] Error 1
CMakeFiles/Makefile2:1126: recipe for target 'examples/record-playback/CMakeFiles/rs-record-playback.dir/all' failed
make[2]: *** [examples/record-playback/CMakeFiles/rs-record-playback.dir/all] Error 2
CMakeFiles/Makefile2:1133: recipe for target 'examples/record-playback/CMakeFiles/rs-record-playback.dir/rule' failed
make[1]: *** [examples/record-playback/CMakeFiles/rs-record-playback.dir/rule] Error 2
Makefile:424: recipe for target 'rs-record-playback' failed
make: *** [rs-record-playback] Error 2
```
### 3. Another trying for `rs-record`
I think this program also doesn't work well, cause I see couldn't any windows and GUI while it is runing. Just a sentence is telling me the recording time in console.
```
Recording t = 9s
Finished

Process finished with exit code 15
```

Is this normal?

chrisrogers3d, 2020-01-18:
make sure you have a real usb3.0 cable.  the cable that came with mine would only do 6fps because its a usb2 cable.  the realsense viewer will tell you which cable you have.  get usb3.1 (10Gbs)even better than 3.0(5gbs)

YuzheHao, 2020-01-21:
Hi everyone, I find a way to solve this low-fps situation:

I recommend to use real sense-viewer to get the .bag file, then use a .bag decoder to extract the frame data from this .bag file. The decoder could be get from (wiki.ros.org/rqt_bag). We could get all frames (RGB,D,Sensor,TIMESTAMP) with this way.

If your realsense-viewer is also have an extremely low-fps, please try to uninstall all the realsense packages and left realsense-viewer only. I don't know why this works, but my realsense-viewer becomes being able to record normal 30-fps RGB+D+Sensor .bag video after doing this. Maybe some packages are overlapped?

I hope this could be helpful.

If have any other question, please reopen this issue

closed 2020-01-21 by YuzheHao