realsenseai/librealsense #5596: How to get the current exposure value of the camera auto exposure mode with code?

Dontla, 2020-01-07:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435| 
| Firmware Version                   | 05.11.06.250| 
| Operating System & Version |  Win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.25.0                     | 
| Language                            |  python3.6                           | 
| Segment			|  Robot                       | 

### Issue Description
<Describe your issue / question / feature request / etc..>
After the camera is automatically exposed, I want to increase or decrease the value of the current exposure based on the image effect. How can I do this?
Best regard

radfordi, 2020-01-08:
```C++
long long get_exposure_time(rs2::frame f) {
    if (f.supports_frame_metadata(RS2_FRAME_METADATA_ACTUAL_EXPOSURE))
        return f.get_frame_metadata(RS2_FRAME_METADATA_ACTUAL_EXPOSURE);
    else
        return 0; // unknown
```

Dontla, 2020-01-10:
@radfordi But when I try to get the actual_exposure parameter of color_frame, it prompts me RuntimeError: metadata not available, do you know what's going on?

Isn't the actual_exposure parameter obtained using color_frame?
[image]
[image]

RealSenseCustomerSupport, 2020-01-17:
----------------------------------------------

Hi @Dontla,

Please make sure that you have metadata support enabled in your system. In general, prior to getting the value of metadata attribute you should check whether or not it is supported in your system.In the code snippet provided by @radfordi,line

`if (f.supports_frame_metadata(RS2_FRAME_METADATA_ACTUAL_EXPOSURE))`

may serve as an example of such a check.

Thank you!

radfordi, 2020-01-17:
If I understand @Dontla's question correctly, I would restate it as, under what conditions is the exposure time available for the RGB frames?  Is there some setting that one can set to ensure that the RGB frames store their associated exposure times in their metadata?

The answer is that frame metadata is only supported on very recent Linux kernels or on older kernels if you have patched kernel modules.   The easiest way to get these patched kernel modules is to [`apt-get install librealsense2-dkms`](https://github.com/IntelRealSense/librealsense/blob/master/doc/distribution_linux.md).   This is covered in the [frame metadata documentation](https://github.com/IntelRealSense/librealsense/blob/master/doc/frame_metadata.md), but it should really be mentioned in the error messages that are thrown when you try and access metadata like you did above.

RealSenseCustomerSupport, 2020-01-22:
----------------------------------------------

@Dontla,

Do you need any additional help with this question?

Dontla, 2020-01-23:
After I updated Intel Realsense SDK or UWP, it can get the metadata normally, but I do n’t know why, this is a world mystery to me ...

closed 2020-01-28 by RealSenseCustomerSupport

PierreLouisT, 2022-06-03:
Hello ! Do you know if it is possible to get the current value of luxmeter. I really need to quantify the quantity of light which comes to the camera outside. I could use exposure but I dont have any exposure time so it is complicated.. 
If it is not possible, do you have any other suggestion ? By the way how can I get the current exposure value just in case ? I tried the differents ideas above without any success :) 

Thank you for all the help you could bring me,