realsenseai/librealsense #5598: Unable to install kernel patch for ubuntu 18.04 kernel 5.0.0

emmanuel-senft, 2020-01-07:
| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       |  SR300  | 
| Operating System & Version |  Linux (Ubuntu 18.04) | 
| Kernel Version (Linux Only)    |  (5.0.0-37-generic)                                         | 
| Platform                                 | PC  |
| SDK Version                            |  2.31.0                           | 
| Segment			|   Robot | 

### Issue Description
I am on the LTS version of Ubuntu 18.04 with kernel 5.0.0-37-generic, and I have tried to install librealsense both with sudo apt and from source, and I cannot patch the kernel.
With apt, I have a crash report with this error: 
scripts/Makefile.build:42: /var/lib/dkms/librealsense2-dkms/1.3.10/build/5.0.0/drivers/media/usb  /uvc/Makefile: No such file or directory
And from source, the command:
./scripts/patch-realsense-ubuntu-lts.sh
fails with: 
Replacing videodev  -
Applying the patched module ... modprobe: ERROR: could not insert 'videodev': Operation not permitted
Failed to insert the patched module. Operation is aborted, the original module is restored
Verify that the current kernel version is aligned to the patched module version

Could you please indicate me if I missed a step or if the procedure for kernel 5.0.0 is different?

Thank you in advance.

ev-mp, 2020-01-14:
@emmanuel-senft hello, what does the kernel log report when executing `./scripts/patch-realsense-ubuntu-lts.sh` ?
Is it possible that you have Secure Boot turned on? #5231

emmanuel-senft, 2020-01-16:
I did have the Secure Boot turned on.. That would explain the issue.
Thank you!

closed 2020-01-16 by emmanuel-senft