realsenseai/librealsense #5599: Matlab librealsense_mex not recognising commands after first run

rlav440, 2020-01-07:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i  | 
| Firmware Version                   | 05.11.15.00 | 
| Operating System & Version |   Win 10| 
| Platform                                 | PC |
| SDK Version                            |  2.0.0                       | 
| Language                            |  MATLAB   2018b                    | 
| Segment			|  Academic                          | 
----------------------------------------------------------------------------------------------------

### Issue Description

Using the MATLAB Wrapper for the SDK, the MEX library will frequently be unable to find any of the included functions after an initial first run (causing the following error or equivalents).

> Error using librealsense_mex
Unknown Command received.
Error in realsense.pointcloud (line 8)
out = realsense.librealsense_mex('rs2::pointcloud', 'new');

While the **depth_example()** code (and similar code examples that grab a single frame then stop the pipeline) can be executed multiple times, attempting to implement a stream similar to the "capture" C++ example will initially run successfully, then any subsequent attempt will be met with the above error.

It is suspected that this is aggravated as MATLAB cannot unload shared libraries from memory (even when clearing the associated .mex) and the only way to get the code to function again is to fully restart MATLAB.

[CodeExample.zip](https://github.com/IntelRealSense/librealsense/files/4032715/CodeExample.zip)

rlav440, 2020-01-12:
For further details:
Attached code can fail to start (i.e, the command to set up a new config is not recognized) even if **depth_example()** has been run successfully multiple times beforehand.

RealSenseCustomerSupport, 2020-02-03:
----------------------------------------------

Had you tried not create cfg and pipe handler inside the init-pipeline function ( i.e, create cfg and pipe in the main script continuousFitting.m )to see if that can help to mitigate the issue you found?  if you found the handler is not return correctly when init-pipeline is called

connelld92, 2020-03-13:
I ran into this same problem and after a bunch of debugging it looks like you have to explicitly call the destructor for the frame objects as well as the pipe object.
For example if i create objects for the frame like this : 
     frame = pipe.wait_for_frames();
    depthFrame = frame.get_depth_frame();
    colorFrame = frame.get_color_frame();

Call the destructor before releasing the pipe object 

delete(depthFrame);
delete(colorFrame);
delete(frame);

This stopped me from having to restart MATLAB to use the wrapper again.

HTH,
Connell

RealSenseCustomerSupport, 2020-03-23:
----------------------------------------------

Hi rlav440,
could solution/suggestion provided so far can help on your case?

rlav440, 2020-04-03:
Sorry for the delay in commenting on this: I swapped to the python wrapper as part of a movement away from Matlab and this issue was no more. I don't have access to the setup I was previously using because of current events, but once I do I will verify connelld92's suggestion.

closed 2020-08-24 by RealSenseCustomerSupport

kossyprg, 2021-08-02:
I started to use D435 camera 4 days ago and had a same problem("Unknown Command received.").
If I removed **"clear all"** command, the error was solved although I don't know why.
I hope this will help the debugging.