realsenseai/librealsense #5602: Where can I find the values that can be set for each parameter in pyrealsense2 \ option.py?

Dontla, 2020-01-08:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435| 
| Firmware Version                   | 05.11.06.250| 
| Operating System & Version |  Win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.25.0                     | 
| Language                            |  python3.6                           | 
| Segment			|  Robot                       | 

### Issue Description
<Describe your issue / question / feature request / etc..>
In pyrealsense2 \ option.py, there are many parameters that can be set, but I have not found what values they can be set to. Can you help me find them
[image]

Best regard

radfordi, 2020-01-08:
The [possible values for `RS2_OPTION_AUTO_EXPOSURE_MODE`](https://github.com/IntelRealSense/librealsense/blob/master/src/ds5/ds5-motion.cpp#L265-L271) seem to be these:
```C++
        ep->register_option(RS2_OPTION_AUTO_EXPOSURE_MODE,
                                std::make_shared<auto_exposure_mode_option>(auto_exposure,
                                                                            ae_state,
                                                                            option_range{0, 2, 1, 0},
                                                                            std::map<float, std::string>{{0.f, "Static"},
                                                                                                         {1.f, "Anti-Flicker"},
                                                                                                         {2.f, "Hybrid"}}));
```

RealSenseCustomerSupport, 2020-02-04:
----------------------------------------------

@radfordi  Please refer to below links to get the value range. Any other questions? Thanks!

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/pybackend.cpp#L101-L165
https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/pyrs_options.cpp#L13

RealSenseCustomerSupport, 2020-02-13:
----------------------------------------------

@radfordi  Any other questions for this ticket? Looking forward to your reply. Thanks!

RealSenseCustomerSupport, 2020-02-25:
----------------------------------------------

@radfordi Will close the ticket if no other questions. Thanks!

closed 2020-02-25 by RealSenseCustomerSupport