realsenseai/librealsense #5608: rs2::pipeline start() fails when D435i accelerometer stream is enabled

swack-cybernet, 2020-01-08:
----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435i
| Firmware Version                   | 05.12.01.00 | 
| Operating System & Version |   Linux (Ubuntu 18.04.3)   | 
| Kernel Version (Linux Only)    |  4.9.140-tegra+                                        | 
| Platform                                 | Jetson TX2  |
| SDK Version                            |  2.31.0                          | 
| Language                            |  C++/OpenCV                         | 
| Segment			|  Others                         | 

### Issue Description
I've written some code to do real-time recording and rendering of RGB, Depth, and IMU frames coming off my D435i. 

At the top of the main:
```c++
rs2::config cfg;
rs2::pipeline csc_rs2_pipeline;
cfg.enable_stream(RS2_STREAM_ACCEL, -1, 0);

rs2::pipeline_profile profileStreams;

try {
	profileStreams = csc_rs2_pipeline.start(cfg, frameCallback);
}
catch(std::exception exc)
{
	std::cout << "Pipeline issue: " << exc.what() << std::endl;
}
```

The callback:
```
void configureStreamCallbacks(const rs2::frame& frame)
{
	std::cout << ".";
}
```

This worked fine in my build of Windows 10 (Pro, Version: 1809, OS build: 17763.914) with RealSense SDK version 2.29.0, but in bringing it over to my Jetson TX2, I found that it was no longer working properly. By disconnecting and reconnecting the camera's USB, I can get it to work once, but no subsequent runs work. Seems like it could be getting into a bad state somehow.

The problem seems to lie in the pipeline start() function override that takes an rs2::config and a frame callback definition failing when I have the IMU's accelerometer stream enabled. No periods appear in the console, leading me to believe that the callbacks are not even getting started.

The dmesg log for this case is as follows:
```
Jan  8 15:37:51 nvidia-desktop kernel: [91060.220303] usb 2-2: new SuperSpeed USB device number 24 using tegra-xusb
Jan  8 15:37:51 nvidia-desktop kernel: [91060.241269] usb 2-2: New USB device found, idVendor=8086, idProduct=0b3a
Jan  8 15:37:51 nvidia-desktop kernel: [91060.241287] usb 2-2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Jan  8 15:37:51 nvidia-desktop kernel: [91060.241298] usb 2-2: Product: Intel(R) RealSense(TM) Depth Camera 435i
Jan  8 15:37:51 nvidia-desktop kernel: [91060.241308] usb 2-2: Manufacturer: Intel(R) RealSense(TM) Depth Camera 435i
Jan  8 15:37:51 nvidia-desktop kernel: [91060.241318] usb 2-2: SerialNumber: 925223050981
Jan  8 15:37:51 nvidia-desktop kernel: [91060.248940] uvcvideo: Unknown video format 00000050-0000-0010-8000-00aa00389b71
Jan  8 15:37:51 nvidia-desktop kernel: [91060.248998] uvcvideo: Unknown video format 00000032-0000-0010-8000-00aa00389b71
Jan  8 15:37:51 nvidia-desktop kernel: [91060.249166] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
Jan  8 15:37:51 nvidia-desktop kernel: [91060.259264] uvcvideo: Unable to create debugfs 2-24 directory.
Jan  8 15:37:51 nvidia-desktop kernel: [91060.259558] uvcvideo 2-2:1.0: Entity type for entity Intel(R) RealSense(TM) Depth Ca was not initialized!
Jan  8 15:37:51 nvidia-desktop kernel: [91060.270171] uvcvideo 2-2:1.0: Entity type for entity Processing 2 was not initialized!
Jan  8 15:37:51 nvidia-desktop kernel: [91060.278642] uvcvideo 2-2:1.0: Entity type for entity Camera 1 was not initialized!
Jan  8 15:37:51 nvidia-desktop kernel: [91060.286782] input: Intel(R) RealSense(TM) Depth Ca as /devices/3530000.xhci/usb2/2-2/2-2:1.0/input/input26
Jan  8 15:37:51 nvidia-desktop kernel: [91060.288413] uvcvideo: Unknown video format 36315752-1a66-a242-9065-d01814a8ef8a
Jan  8 15:37:51 nvidia-desktop kernel: [91060.288426] uvcvideo: Found UVC 1.50 device Intel(R) RealSense(TM) Depth Camera 435i (8086:0b3a)
Jan  8 15:37:51 nvidia-desktop kernel: [91060.303291] uvcvideo: Unable to create debugfs 2-24 directory.
Jan  8 15:37:51 nvidia-desktop kernel: [91060.303468] uvcvideo 2-2:1.3: Entity type for entity Processing 7 was not initialized!
Jan  8 15:37:51 nvidia-desktop kernel: [91060.311502] uvcvideo 2-2:1.3: Entity type for entity Extension 8 was not initialized!
Jan  8 15:37:51 nvidia-desktop kernel: [91060.319464] uvcvideo 2-2:1.3: Entity type for entity Camera 6 was not initialized!
Jan  8 15:37:51 nvidia-desktop mtp-probe: checking bus 2, device 24: "/sys/devices/3530000.xhci/usb2/2-2"
Jan  8 15:37:51 nvidia-desktop mtp-probe: bus: 2, device: 24 was not an MTP device
Jan  8 15:37:51 nvidia-desktop /usr/lib/gdm3/gdm-x-session[6191]: (II) config/udev: Adding input device Intel(R) RealSense(TM) Depth Ca (/dev/input/event5)
Jan  8 15:37:51 nvidia-desktop /usr/lib/gdm3/gdm-x-session[6191]: (**) Intel(R) RealSense(TM) Depth Ca: Applying InputClass "libinput keyboard catchall"
Jan  8 15:37:51 nvidia-desktop /usr/lib/gdm3/gdm-x-session[6191]: (II) Using input driver 'libinput' for 'Intel(R) RealSense(TM) Depth Ca'
Jan  8 15:37:51 nvidia-desktop /usr/lib/gdm3/gdm-x-session[6191]: (II) systemd-logind: got fd for /dev/input/event5 13:69 fd 56 paused 0
Jan  8 15:37:51 nvidia-desktop /usr/lib/gdm3/gdm-x-session[6191]: (**) Intel(R) RealSense(TM) Depth Ca: always reports core events
Jan  8 15:37:51 nvidia-desktop /usr/lib/gdm3/gdm-x-session[6191]: (**) Option "Device" "/dev/input/event5"
Jan  8 15:37:51 nvidia-desktop /usr/lib/gdm3/gdm-x-session[6191]: (**) Option "_source" "server/udev"
Jan  8 15:37:51 nvidia-desktop /usr/lib/gdm3/gdm-x-session[6191]: (II) event5  - Intel(R) RealSense(TM) Depth Ca: is tagged by udev as: Keyboard
Jan  8 15:37:51 nvidia-desktop /usr/lib/gdm3/gdm-x-session[6191]: (II) event5  - Intel(R) RealSense(TM) Depth Ca: device is a keyboard
Jan  8 15:37:51 nvidia-desktop /usr/lib/gdm3/gdm-x-session[6191]: (II) event5  - Intel(R) RealSense(TM) Depth Ca: device removed
Jan  8 15:37:51 nvidia-desktop /usr/lib/gdm3/gdm-x-session[6191]: (**) Option "config_info" "udev:/sys/devices/3530000.xhci/usb2/2-2/2-2:1.0/input/input26/event5"
Jan  8 15:37:51 nvidia-desktop /usr/lib/gdm3/gdm-x-session[6191]: (II) XINPUT: Adding extended input device "Intel(R) RealSense(TM) Depth Ca" (type: KEYBOARD, id 8)
Jan  8 15:37:51 nvidia-desktop /usr/lib/gdm3/gdm-x-session[6191]: (**) Option "xkb_model" "pc105"
Jan  8 15:37:51 nvidia-desktop /usr/lib/gdm3/gdm-x-session[6191]: (**) Option "xkb_layout" "us"
Jan  8 15:37:51 nvidia-desktop /usr/lib/gdm3/gdm-x-session[6191]: (WW) Option "xkb_variant" requires a string value
Jan  8 15:37:51 nvidia-desktop /usr/lib/gdm3/gdm-x-session[6191]: (WW) Option "xkb_options" requires a string value
Jan  8 15:37:51 nvidia-desktop /usr/lib/gdm3/gdm-x-session[6191]: (II) event5  - Intel(R) RealSense(TM) Depth Ca: is tagged by udev as: Keyboard
Jan  8 15:37:51 nvidia-desktop /usr/lib/gdm3/gdm-x-session[6191]: (II) event5  - Intel(R) RealSense(TM) Depth Ca: device is a keyboard
Jan  8 15:37:53 nvidia-desktop kernel: [91062.447792] usb 2-2: usb_suspend_both: status 0
Jan  8 15:37:53 nvidia-desktop kernel: [91062.448122] usb usb2: usb_suspend_both: status 0
Jan  8 15:38:14 nvidia-desktop /usr/lib/gdm3/gdm-x-session[6191]: (II) config/udev: removing device Intel(R) RealSense(TM) Depth Ca
Jan  8 15:38:14 nvidia-desktop /usr/lib/gdm3/gdm-x-session[6191]: (**) Option "fd" "56"
Jan  8 15:38:14 nvidia-desktop /usr/lib/gdm3/gdm-x-session[6191]: (II) event5  - Intel(R) RealSense(TM) Depth Ca: device removed
Jan  8 15:38:14 nvidia-desktop /usr/lib/gdm3/gdm-x-session[6191]: (II) UnloadModule: "libinput"
Jan  8 15:38:14 nvidia-desktop /usr/lib/gdm3/gdm-x-session[6191]: (II) systemd-logind: releasing fd for 13:69
Jan  8 15:38:14 nvidia-desktop /usr/lib/gdm3/gdm-x-session[6191]: (--) NVIDIA(GPU-0): DELL E2211H (DFP-0): connected
Jan  8 15:38:14 nvidia-desktop /usr/lib/gdm3/gdm-x-session[6191]: (--) NVIDIA(GPU-0): DELL E2211H (DFP-0): External TMDS
Jan  8 15:38:44 nvidia-desktop kernel: [91113.555813] usb 2-2: usb_suspend_both: status 0
Jan  8 15:38:44 nvidia-desktop kernel: [91113.556131] usb usb2: usb_suspend_both: status 0

```

Some help would be greatly appreciated!

dorodnic, 2020-01-09:
Hi @swack-cybernet 
Are you using debian packages for Jetson or building the SDK from source?

swack-cybernet, 2020-01-09:
Hi @dorodnic 
I built the SDK from source.

dorodnic, 2020-01-09:
Please give the Debian version a try, it is accessing IMU in slightly different way that we believe to be more robust

swack-cybernet, 2020-01-09:
Installed the debian packages:
```
librealsense2/bionic,now 2.31.0-5ubuntu5~1.gbp95d195 arm64 [installed,automatic]
librealsense2-dbg/bionic 2.31.0-5ubuntu5~1.gbp95d195 arm64
librealsense2-dev/bionic,now 2.31.0-5ubuntu5~1.gbp95d195 arm64 [installed]
librealsense2-gl/bionic,now 2.31.0-5ubuntu5~1.gbp95d195 arm64 [installed,automatic]
librealsense2-gl-dbg/bionic 2.31.0-5ubuntu5~1.gbp95d195 arm64
librealsense2-gl-dev/bionic 2.31.0-5ubuntu5~1.gbp95d195 arm64
librealsense2-udev-rules/bionic,now 2.31.0-5ubuntu5~1.gbp95d195 arm64 [installed,automatic]
librealsense2-utils/bionic,now 2.31.0-5ubuntu5~1.gbp95d195 arm64 [installed]

```
Unfortunately, I still have the same problem.

swack-cybernet, 2020-03-06:
A very gentle bump on this. The issue still persists unfortunately.

RealSenseCustomerSupport, 2020-06-11:
----------------------------------------------

Hi,

Are you still experiencing issues with this on latest. We are able to run the sample code you provided and successfully get callbacks with or without ACCEL stream enabled on an x86 NUC and arm64 Jetson.

Have you tried just installing the sdk packages using these instructions?

https://github.com/IntelRealSense/librealsense/blob/master/doc/installation_jetson.md

I suggest removing any traces of lrs first using:

dpkg -l | grep "realsense" | cut -d " " -f 3 | xargs sudo dpkg --purge

Thanks

RealSenseSupport, 2020-06-22:
Hi @swack-cybernet 

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

swack-cybernet, 2020-06-23:
Hi!

Updating to release 32 and up seems to have fixed the problem. Thank you for all your help!

RealSenseSupport, 2020-07-07:
Glad that resolved your problem. We'll close this issue then. Feel free to re-open / create new if you experience further problems.

closed 2020-07-07 by RealSenseSupport