realsenseai/librealsense #5610: incompatible function arguments for set_option

bfouts-kindred, 2020-01-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                              | D435                                                               | 
| Firmware Version                         | 05.11.15.00                                                      | 
| Operating System & Version       |   Ubuntu 16.04                                               | 
| Kernel Version (Linux Only)         |  4.8.0-36-generic                                         | 
| Platform                                        | PC                                                                   |
| SDK Version                                 |  2.31.0.1235                                                    | 
| Language                                     |  python                                                            | 
| Segment			              |  Robot                                                             | 

### Issue Description
I am trying to run the export_ply_example.py script included in this repository, but it is complaining about the `set_option` call linked below

https://github.com/IntelRealSense/librealsense/blob/master/wrappers/python/examples/export_ply_example.py#L40

I get the following error:
```
Traceback (most recent call last):
  File "generate_ply.py", line 40, in <module>
    ply.set_option(rs.save_to_ply.option_ply_binary, False)
TypeError: (): incompatible function arguments. The following argument types are supported:
    1. () -> pyrealsense2.pyrealsense2.option
```

The script works if I remove both of the `set_option` calls (though the resulting point cloud doesn't seem accurate, which may be unrelated).

These pages don't produce any search results:
https://intelrealsense.github.io/librealsense/python_docs/search.html?q=option_ply_binary
https://intelrealsense.github.io/librealsense/python_docs/search.html?q=option_ply_normals

Has this script fallen out of date? How can I find the options that this will accept?

nirandiw, 2020-01-10:
I have the same issue. The resulting point cloud is very different to the point cloud I get if I export the point cloud via realsense-viewer. Any update?

nauti16, 2020-03-12:
I have the same issue with original example code. Any update?
There are little bit differect logs between 2.33.1 and 2.31.0

1. pyreansense2 version '2.33.1' in windows notebook
---> 38     ply.set_option(rs.pyrealsense2.save_to_ply.option_ply_binary, False)
     39     ply.set_option(rs.pyrealsense2.save_to_ply.option_ply_normals, True)

TypeError: (): incompatible function arguments. The following argument types are supported:
    1. () -> pyrealsense2.pyrealsense2.option

Invoked with: <class 'pyrealsense2.pyrealsense2.save_to_ply'>

2. pyreansense2 version '2.31.0' in Nvidia TX2
---> 33     ply.set_option(rs.save_to_ply.option_ply_binary, False)
     34     ply.set_option(rs.save_to_ply.option_ply_normals, True)
     35 

TypeError: (): incompatible function arguments. The following argument types are supported:
    1. () -> pyrealsense2.option

Invoked with: <class 'pyrealsense2.save_to_ply'>

RealSenseSupport, 2020-07-23:
Hi @bfouts-kindred 

Are you still experiencing this? We cannot reproduce this issue on the latest librealsense ver 2.36.0. The sample generates 1.ply and can be read successfully by supported software. Also the set_options works as well and are defined here:

https://github.com/IntelRealSense/librealsense/blob/f7cdf6e8961e1709f6d864bdb33095c00a671ca7/wrappers/python/python.cpp#L33

Thanks

closed 2020-08-14 by RealSenseSupport