realsenseai/librealsense #5611: Option.auto_exposure_mode parameter in option.py in pyrealsense belongs to which sensor in D435?

Dontla, 2020-01-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | D435| 
| Firmware Version                   | 05.11.06.250| 
| Operating System & Version |  Win10  | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC  |
| SDK Version                            | 2.25.0                     | 
| Language                            |  python3.6                           | 
| Segment			|  Robot                       | 

### Issue Description
<Describe your issue / question / feature request / etc..>
I'm curious, I want to print the description of this parameter, but it can't print, suggesting that it may not belong to the color sensor of D435, which sensor should it belong
Best regard

`get_option_description = sensor.get_option_description(rs.option.auto_exposure_mode)
print(get_option_description)`

RuntimeError: object doesn't support option #36

RealSenseSupport, 2020-12-05:
Hi,

Will you be needing further help with this? If we don’t hear from you in 7 days, this issue will be closed.

Thanks

closed 2020-12-14 by RealSenseSupport

PierreLouisT, 2022-08-30:
Hello, I work with D455 realsense camera, I would like your advices. I read all of this document which tells what are the differents data SDK as accuracy, projector temperature, etc.. However, Some of data do not work using pyrealsense, 
for example, in my code below, I can have the gain but not saturation using get_options
[image]

Hope you'll know what is my problem.. 
By the way do you know if I can get a more precise documentation, there are not description of each parameters and some of them are unknown for me (for example, Asic temperature or mc temperature, don't know what that is) 
Thank you for your help, 

PL