realsenseai/librealsense #5612: Multicam synchronization without HW connections

yanxh97, 2020-01-09:
* Before opening a new issue, we wanted to provide you with some useful suggestions (Click "Preview" above for a better view):

    * Consider checking out SDK [examples](https://github.com/IntelRealSense/librealsense/tree/master/examples#sample-code-for-intel-realsense-cameras).
    * Have you looked in our [documentations](https://github.com/IntelRealSense/librealsense/tree/master/doc#useful-links)?
    * Is you question a [frequently asked one](https://github.com/IntelRealSense/librealsense/wiki/Troubleshooting-Q%26A)?
    * Try [searching our GitHub Issues](https://github.com/IntelRealSense/librealsense/issues?utf8=%E2%9C%93&q=is%3Aissue) (open and closed) for a similar issue.

* All users are welcomed to report bugs, ask questions, suggest or request enhancements and generally feel free to open new issue, even if they haven't followed any of the suggestions above :)

----------------------------------------------------------------------------------------------------

| Required Info                         |                                                                |
|---------------------------------|------------------------------------------- |
| Camera Model                       | {  D400 } | 
| Firmware Version                   | (Open RealSense Viewer --> Click info) | 
| Operating System & Version |   {Linux (Ubuntu 14/16/17)   | 
| Kernel Version (Linux Only)    |  (e.g. 4.14.13)                                         | 
| Platform                                 | PC/Raspberry Pi/ NVIDIA Jetson / etc..  |
| SDK Version                            |  { legacy / 2.<?>.<?> }                          | 
| Language                            |  {C++ }                          | 
| Segment			|  {Robot/Smartphone/VR/AR/others }                          | 

### Issue Description
I'm working with multi(3) D435s and the synchronization puzzles me. For the task is not that demanding in aspect of synchronization,  I would prefer carrying it out without HW sync. Then which timestamp should I check when aligning framesets from different source device? Is there any tricks about the poll_for_frame and wait_for_frame method over this problem? 
What I've tried: Suppose the answer to the 1st question is back-end timestamp, I select a master cam and wait_for_frame of it at the start of every loop, and then use poll_for_frame to update the slaves(if it returns false, use the frameset of the former loop/cycle). If the time interval seems too long, I would wait_for_frame of one of the slaves.  (for example, the stream is 30Hz and if the frameset I poll comes over 20 ms earlier than master, I would  call wait_for_frame, so the slave's frameset comes later but with shorter time interval)

MartyG-RealSense, 2020-01-09:
The subject of 'software sync' without hardware is discussed in the link below.

https://github.com/IntelRealSense/librealsense/issues/2148

yanxh97, 2020-01-10:
Thanks. @MartyG-RealSense
To be more specific, what function/metadata should I call to get the host timestamp caculated from `RS2_FRAME_METADATA_BACKEND_TIMESTAMP - (RS2_FRAME_METADATA_FRAME_TIMESTAMP - RS2_FRAME_METADATA_SENSOR_TIMESTAMP)) `

MartyG-RealSense, 2020-01-10:
My interpretation of the very last comment by @dorodnic in the discussion that I linked to is that the RealSense developers incorporated the formula that you quoted into the SDK from version 2.22.0 onwards as.a feature called Global Camera Time.

https://github.com/IntelRealSense/librealsense/pull/3909

RealSenseCustomerSupport, 2020-01-17:
----------------------------------------------

Hi @yanxh97 Do you need further help on this issue?Thanks

closed 2020-01-28 by RealSenseCustomerSupport